PX4/PX4-Autopilot #8147: contents of extras.txt not executed upon boot-up of pixhawk mini

cs147, 2017-10-17:
Hello all,
I am using a Pixhawk Mini. On `/fs/microsd/extras.txt', I put `mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 200` (as specified [here](https://dev.px4.io/en/advanced/system_startup.html))

When the board boots up, the IMU rate is 50 hz. If I type this into the command line of the system console after the Pixhawk Mini boots up, the IMU rate is increased. So the contents of `extras.txt` are not being executed. Is this a boot-up error? How can I debug this?

bkueng, 2017-10-18:
Note that the path is `/fs/microsd/etc/extras.txt`, not `/fs/microsd/extras.txt`.

cs147, 2017-10-18:
@bkueng Sorry I typed it in wrong here. Yes, it is in `/etc` folder

SubMishMar, 2017-10-18:
Where exactly is the etc folder?

cs147, 2017-10-18:
@SubMishMar `/fs/microsd/etc/`

SubMishMar, 2017-10-18:
Thanks! but my sd card is empty!

bkueng, 2017-10-20:
@cs147 I just tested on a Pixracer, and the command is executed. Are you using Windows to edit the file? If so make sure that you use Unix line-endings.
Alternatively you can create the file directly on the system console:
```
cd /fs/microsd/etc
echo "mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 200" > extras.txt
```

@SubMishMar you need to create the folder `etc` on the SD card

closed 2017-10-20 by LorenzMeier

cs147, 2017-10-30:
@LorenzMeier This issue is not closed

@bkueng I did exactly what you said by creating the file directly on the system cosnole. It still does not get executed upon startup