PX4/PX4-Autopilot #8148: Reboot via QGC does not restore dropped RC connection

JonReacher, 2017-10-18:
Quite often I have the RC control stop working after powering on the autopilot.  It doesn't lose signal or indicate lost connection in QGC, it just stops responding to inputs from the RC radio. This only happens before arming, never after. It seems to not happen if I constantly move the sticks pretty regularly, but if left alone for a minute.

I'm not sure why this happens, but at the least if I could reboot via QGC and restore control that would be fine. That doesn't work however, so I have to open a hatch and physically unplug the power. Is there a known reason why the autopilot stops responding to input? If it's a big process to fix the issue, is there a way to restore it via software reboot for the short term?

dagar, 2017-10-20:
Could you describe your hardware? Particular board, RC equipment, etc.

I'm not sure what the problem is yet, but I'd be concerned there's something more fundamentally wrong if a software reboot isn't fixing it.

JonReacher, 2017-10-20:
Pixhawk 1 (2.4.6 from Holybro), FrSky Taranis XD9 plus and X8R receiver. I would think it's a popular combo. Let me know if you need me to try anything with this setup to debug.

dagar, 2017-10-22:
Yes that is a popular combination, which is one reason I suspect some kind of hardware problem.
Can you post any corresponding logs? I'm not sure if they'll show anything, but it's a good place to start.

JonReacher, 2017-10-23:
I spent the past couple days really trying to get this figured out. The reason I posted before doing this is because I was thinking that there must be a known issue with a fix or workaround that I wasn't finding.

I have a log here: https://logs.px4.io/plot_app?log=a0c705fa-0c94-41e8-807f-8ad10fb8a33e

After the manual control lost event, the rc receiver is still indicating a signal. There is no indication from the radio or receiver of any signal loss at any point. Turning the radio off and back on does not restore manual control, neither does a soft reboot.

 Also, I'm not sure why it only happens if it sits with no input for varying amounts of time. Any input seems to 'reset' the clock, even though it's not consistent about when it happens. It's also never happened during a flight, with some flights of 30-45 minutes.

closed 2018-02-21 by PX4BuildBot