PX4/PX4-Autopilot #8151: Posctrl Z position hold with LIDAR and PX4Flow

siddyyagno3, 2017-10-19:
The Quadrotor has a optical flow px4 flow and Lidar lite V3(i2c). While flying manually the drone is performing normally even when switching to Altitude Hold mode the altitude hold is good. If I put my hand just below the  lidar drone goes up and comes down to actual height.  
Subsequently, while testing position control mode it is holding the position earlier there were oscillations but after tuning it is pretty stable and stays at the position without GPS. But when I try to put my hand below the sensor the drone goes up and for some second it stays there and then it start to comes down but then it actually goes up.
Reviewing the flight logs I observed that Z setpoint changes to -6 instead of -2 and drone somewhat stays there and we have to bring it down. And this thing is repeatable. It is the test case if a car comes below it  should go up and come back to its original altitude. 
Is this z setpoint change is as expected or some bad parameter tuning or worse some bug in firmware?
Px4 Flight logs:
https://logs.px4.io/plot_app?log=2e693261-f0df-4f75-867b-4ac39b73f19c

siddyyagno3, 2017-10-20:
I again tested the flight with GPS on EKF2-aid-mask set to 3. Similar, issue happened. 
[Local position z](https://logs.px4.io/plot_app?log=8d67494e-663c-428a-a1b3-1b1d73a93311#Nav-Local-Position-Z)
I realised for small distance reading change posctrl is good i.e. it comes back down to its position but if estimated position crosses 0 or becomes positive it changes its set point near to -5.5 or so. Also estimated position shouldn't  cross 0  as Distance reading is good for low values and it is not becoming negative.
@priseborough  @ChristophTobler

ChristophTobler, 2017-10-20:
Seems like the ekf was deadreckoning for too long and then reset the height.
[image]
By that time the z estimate has already diverged too much and that stays after the reset.

ChristophTobler, 2017-10-25:
https://github.com/PX4/Firmware/pull/8183 is a fix (partly). With this one your vehicle would stay at ~3m after the reset instead of applying the delta between estimated z and setpoint z.
The other issue is that you have quite a lot of vibrations and that's why your z estimate goes the wrong direction -> resulting in a large delta.

closed 2017-10-27 by ChristophTobler