PX4/PX4-Autopilot #8161: mathlib + matrix

dagar, 2017-10-20:
We currently have duplicated libraries for Matrix, Vector, Quaternions across mathlib and matrix. This is unnecessarily confusing for adopters.

https://github.com/PX4/Firmware/tree/master/src/lib/mathlib/math
https://github.com/PX4/Matrix/tree/e595ebb9a704c24aeae990dac768d26949fcaee0/matrix

We should discuss a path forward to remove the duplication. There's also the matrix submodule being included twice https://github.com/PX4/Firmware/issues/7991.

Since ecl is a core dependency I think Firmware should use the version of Matrix included by ecl. Then the existing mathlib can gradually be gutted so that it simply uses Matrix + any functions needed for compatibility. Eventually the need for compatibility functions can be removed entirely and mathlib Vector, Matrix, Quaternion can be a typedef for the Matrix version.

@MaEtUgR @Stifael FYI

MaEtUgR, 2017-10-23:
I use the `matrix` library everywhere I do changes. The problem is there's usually a big bulk to change it for the whole module and that's why I end up with conversions to `math::matrix` back again to stay compatible which is far from nice.

I think we need a joint effort for a release (why not 1.8) to concentrate and switch the majority. They are pretty easy to switch out in most cases because the basic interface is the same but I'm personally against a typdef because it could generate even more confusion in the future and even lead to problems because of incompatibility.

Stifael, 2017-10-25:
@MaEtUgR @dagar how about to switch to matrix any time we encounter vectors/matrices that are of type mathlib, even if it just means to copy it directly over to matrix for just one calculations. This way we can do it incrementally and it is probably easier to switch everything in one shot even if it is just module based

dagar, 2017-10-25:
@Stifael that could work if we make sure everyone is actually working towards that goal. We've had matrix for 2 years and new mathlib matrix usage is still coming in, so I figured it might be time to try something new.

It's messier, but the other compelling part about getting to typdef compatibility is not breaking external code.

dagar, 2017-10-25:
We should push for what @Stifael suggested regardless (replace mathlib matrix usage with Matrix as you come across it). Removing mathlib matrix entirely or replacing with typedefs can be decided separately.

Stifael, 2017-10-26:
for reference: https://github.com/PX4/Firmware/issues/7991

Stifael, 2017-10-26:
@dagar, I would like to start with ecl l1 controller, which currently uses math::Vector. 
As we have discussed, I would like to enable L1 for local coordinates as well, and by doing so, i will switch to matrix

MaEtUgR, 2017-10-30:
I want to start out with the mc attitude controller. I'm anyways looking at it for quaternion control and introduced new matrix library functionality for making that more efficient.

closed 2018-02-21 by PX4BuildBot