PX4/PX4-Autopilot #8163: Possible Lidar "double-correction" bug

elikos-user, 2017-10-20:
Hello,

In the " Firmware/src/modules/local_position_estimator/sensors/lidar.cpp" file there seems to be a possible bug in the "BlockLocalPositionEstimator::lidarMeasure" and "BlockLocalPositionEstimator::lidarCorrect" methods. 

The following pieces of code are repeated twice and seem to do the same job:
```
y(0) = (d + _lidar_z_offset.get()) *
	       cosf(_eul(0)) *
	       cosf(_eul(1));
```

```
if (lidarMeasure(y) != OK) { return; }

	// account for leaning
	y(0) = y(0) *
	       cosf(_eul(0)) *
	       cosf(_eul(1));
```

In the second snippet, the call to the method "lidarMeasure" executes the code in the first snippet. And after the if statement, the first snipped is re-executed. The leaning of the drone is then taken twice into consideration.

Thank you

dagar, 2017-10-25:
Could you take a look at this @TSC21?

TSC21, 2017-10-25:
@elikos-user nice catch. Have you tried to run it with and without the double correction?

TSC21, 2017-10-25:
In any case you are right and I'm going to add a fix. Thanks!

closed 2017-10-25 by TSC21