PX4/PX4-Autopilot #8164: ROS/Camera

alirezaka, 2017-10-21:
Hi
I would like to integrate PX4 in ROS and my main objective is controlling my UAV by pixhawk which receives its commands from a ROS node which basically publishes UAV commands  by the information it gets from a camera linked to robot but since i am using instructions mentioned here ( https://dev.px4.io/en/simulation/gazebo.html ) gazebo runs separately from ROS and I am not able to receive any image in ROS .
Is there any way to have particular PX4 package for ROS ?
Or even a way to receive image via ROS when gazebo runs separately from ros I mean without gazebo_ros package

TSC21, 2017-10-21:
https://github.com/mavlink/mavros

alirezaka, 2017-10-21:
Thanks for your comment but i know that I can connect to px4 via mavros/mavlink, what I meant was that how to run px4 like a ROS node or how to add a camera publishing a rostopic when I am working with px4 in the way that mentioned in dev.px4.io website?

ndepal, 2017-10-21:
The PX4 Firmware repository is set up to be compiled as a catkin package (although it may not be documented well). This allows you to run PX4 on your posix system together with ROS.

Since you write about using a PixHawk, that does not seem to be what you want.
If you do in fact have a UAV which flies with a PixHawk running PX4 and want to send commands to PX4 from ROS running externally (on a companion computer or an offboard computer) mavros is what you want, as @TSC21 suggested.

You will want to create a ROS node that uses camera images to make decisions about what the UAV has to do. How you get your camera images into ROS has nothing to do with PX4.
Once you know what your UAV should do, you can send commands to it through mavros. The dev-guide might be of help: https://dev.px4.io/en/ros/mavros_offboard.html

alirezaka, 2017-10-21:
Thank you ndepal for your perfect response.
As i understood you believe that I can use a UAV in gazebo while I am sending Commands through PX4 and at the same time I can have a camera on it which publishes a rostopic .
Am I right ?
If so , could you please give me more detail on how it is possible to fulfil this objective ?

alirezaka, 2017-10-21:
By saying pixhawk I meant PX4.
 because I am working on a sitl gazebo project and in sitl projects PX4 is equivalent to pixhawk

ndepal, 2017-10-21:
Ok, if you want to do this in simulation, you need to add a camera model that publishes ROS messages. This does not have much to do with PX4 and I can't help you much. The Gazebo tutorial http://gazebosim.org/tutorials?tut=ros_gzplugins&cat=connect_ros#Camera might get you started, otherwise you might be best helped by asking those questions on ROS specific forums.

alirezaka, 2017-10-21:
thank you ndepal for the time you spent on my question .

closed 2018-02-21 by PX4BuildBot

Vuwij, 2018-06-04:
Still with this quesiton. Is there any way we can get mavros to publish a camera message from gazebo? /mavros publishes stuff like /mavros/imu/data etc

ndepal, 2018-06-05:
As stated above, this has nothing to do with PX4 or MAVROS. It's a question about gazebo and ROS. You should ask in the relevant forums.