PX4/PX4-Autopilot #8169: make px4fmu-v2_default fail

aaumar, 2017-10-22:
I am new to this and I just followed from fresh install ubuntu 16.04 from devpx4 site and I got problem when running make px4fmu-v2_default, here is the error that I got

> abdularis@abdularis:~/catkin_ws/src/Firmware$ make px4fmu-v2_default[  1%] Built target ver_gen
[  1%] Built target copy_px4io_bin
[  1%] Built target uorb_headers
[  1%] Built target git_ecl
[  1%] Built target git_nuttx
[  1%] Built target mixer_gen
[  1%] Built target airframes_xml
[  1%] Built target parameters_xml
[  1%] Built target nuttx_configure
[  2%] Built target nuttx_context
[  2%] Built target prebuild_targets
[  2%] Built target nuttx_mm_build
[  2%] Built target nuttx_fs_build
[  2%] Built target nuttx_drivers_build
[  3%] Built target nuttx_binfmt_build
[  3%] Built target nuttx_builtins_send_event
[  4%] Built target nuttx_c_build
[  4%] Built target nuttx_builtins_fmu
[  4%] Built target nuttx_builtins_vtol_att_control
[  5%] Built target nuttx_builtins_mc_pos_control
[  5%] Built target nuttx_builtins_mc_att_control
[  5%] Built target nuttx_builtins_gnd_pos_control
[  5%] Built target nuttx_builtins_ekf2
[  5%] Built target nuttx_builtins_fw_att_control
[  5%] Built target nuttx_arch_build
[  6%] Built target nuttx_builtins_gpio_led
[  6%] Built target nuttx_builtins_navigator
[  6%] Built target nuttx_builtins_ets_airspeed
[  6%] Built target nuttx_builtins_adc
[  6%] Built target nuttx_builtins_land_detector
[  6%] Built target nuttx_builtins_load_mon
[  6%] Built target nuttx_builtins_ulanding_radar
[  6%] Built target nuttx_builtins_ms5525_airspeed
[  6%] Built target nuttx_builtins_ll40ls
[  6%] Built target nuttx_builtins_pwm_input
[  6%] Built target nuttx_builtins_mavlink
[  6%] Built target nuttx_builtins_pwm_out_sim
[  7%] Built target nuttx_builtins_ms5611
[  7%] Built target nuttx_builtins_ms4525_airspeed
[  8%] Built target nuttx_configs_build
[  8%] Built target nuttx_builtins_tone_alarm
[  8%] Built target nuttx_builtins_uorb
[  8%] Built target nuttx_builtins_gnd_att_control
[  8%] Built target nuttx_builtins_commander
[  9%] Built target nuttx_builtins_hmc5883
[  9%] Built target nuttx_builtins_bl_update
[ 10%] Built target nuttx_builtins_sf0x
[ 11%] Built target nuttx_builtins_px4io
[ 12%] Built target nuttx_builtins_camera_feedback
[ 13%] Built target nuttx_builtins_mpu9250
[ 13%] Built target nuttx_builtins_mpu6000
[ 13%] Built target nuttx_builtins_rgbled
[ 13%] Built target nuttx_builtins_teraranger
[ 14%] Built target nuttx_builtins_sdp3x_airspeed
[ 14%] Built target nuttx_builtins_logger
[ 14%] Built target nuttx_builtins_frsky_telemetry
[ 14%] Built target nuttx_builtins_sensors
[ 14%] Built target nuttx_builtins_dataman
[ 14%] Built target nuttx_builtins_hardfault_log
[ 14%] Built target nuttx_builtins_gps
[ 14%] Built target nuttx_builtins_px4flow
[ 14%] Built target nuttx_builtins_camera_trigger
[ 15%] Built target nuttx_builtins_lis3mdl
[ 15%] Built target nuttx_builtins_mixer
[ 16%] Built target nuttx_builtins_config
[ 17%] Built target nuttx_builtins_ver
[ 18%] Built target nuttx_builtins_nshterm
[ 18%] Built target nuttx_builtins_vmount
[ 19%] Built target nuttx_builtins_top
[ 20%] Built target nuttx_builtins_lsm303d
[ 20%] Built target nuttx_builtins_mtd
[ 20%] Built target nuttx_builtins_l3gd20
[ 21%] Built target nuttx_builtins_param
[ 21%] Built target nuttx_builtins_pwm
[ 21%] Built target nuttx_builtins_perf
[ 21%] Built target nuttx_cxx_build
[ 21%] Built target nuttx_builtins_reboot
[ 22%] Built target nuttx_builtins_fw_pos_control_l1
[ 22%] Built target nuttx_sched_build
[ 23%] Built target platforms__common
[ 23%] Built target nuttx_builtins
[ 24%] Built target romfs
[ 24%] Built target platforms__nuttx
[ 39%] Built target uorb_msgs
[ 40%] Built target drivers__device
[ 41%] Built target drivers__stm32__adc
[ 41%] Built target drivers__stm32__tone_alarm
[ 41%] Built target drivers__led
[ 42%] Built target drivers__stm32
[ 42%] Built target drivers__px4fmu
[ 43%] Built target lib__led
[ 44%] Built target drivers__px4io
[ 45%] Built target drivers__boards__px4fmu-v2
[ 45%] Built target drivers__mpu6000
[ 46%] Built target drivers__lsm303d
[ 47%] Built target drivers__l3gd20
[ 50%] Built target drivers__mpu9250
[ 50%] Built target drivers__hmc5883
[ 51%] Built target drivers__ms5611
[ 51%] Built target drivers__sf0x
[ 51%] Built target drivers__teraranger
[ 51%] Built target drivers__ll40ls
[ 52%] Built target drivers__gps
[ 53%] Built target drivers__pwm_out_sim
[ 53%] Built target drivers__ets_airspeed
[ 54%] Built target drivers__airspeed
[ 54%] Built target drivers__ms4525_airspeed
[ 54%] Built target drivers__ms5525_airspeed
[ 55%] Built target drivers__sdp3x_airspeed
[ 56%] Built target drivers__frsky_telemetry
[ 57%] Built target modules__sensors
[ 57%] Built target drivers__px4flow
[ 59%] Built target drivers__vmount
[ 59%] Built target drivers__pwm_input
[ 60%] Built target drivers__lis3mdl
[ 61%] Built target drivers__camera_trigger
[ 61%] Built target drivers__ulanding
[ 61%] Built target systemcmds__bl_update
[ 61%] Built target systemcmds__config
[ 61%] Built target systemcmds__mixer
[ 62%] Built target systemcmds__hardfault_log
[ 62%] Built target systemcmds__mtd
[ 63%] Built target systemcmds__nshterm
[ 63%] Built target systemcmds__perf
[ 63%] Built target systemcmds__param
[ 64%] Built target systemcmds__pwm
[ 64%] Built target systemcmds__reboot
[ 64%] Built target systemcmds__top
[ 65%] Built target systemcmds__ver
[ 66%] Built target modules__load_mon
[ 67%] Built target modules__uORB
[ 69%] Built target modules__commander
[ 69%] Built target modules__gpio_led
[ 72%] Built target modules__navigator
[ 72%] Built target modules__ekf2
[ 73%] Built target modules__camera_feedback
[ 74%] Built target modules__land_detector
[ 74%] Built target lib__external_lgpl
[ 74%] Built target modules__mc_att_control
[ 77%] Built target modules__mavlink
[ 77%] Built target modules__mc_pos_control
[ 80%] Built target lib__ecl
[ 80%] Built target modules__logger
[ 81%] Built target modules__vtol_att_control
[ 82%] Built target modules__systemlib__param
[ 82%] Built target modules__dataman
[ 85%] Built target lib__controllib
[ 86%] Built target modules__systemlib
[ 87%] Built target modules__systemlib__mixer
[ 87%] Built target lib__mathlib
[ 88%] Built target lib__mathlib__math__filter
[ 89%] Built target lib__geo
[ 89%] Built target lib__geo_lookup
[ 90%] Built target lib__conversion
[ 90%] Built target lib__launchdetection
[ 91%] Built target lib__terrain_estimation
[ 91%] Built target lib__runway_takeoff
[ 91%] Built target lib__tailsitter_recovery
[ 91%] Built target lib__version
[ 91%] Built target nuttx_apps_build
[ 91%] Built target platforms__nuttx__px4_layer
[ 92%] Built target df_driver_framework
[ 92%] Built target drivers__rgbled
[ 93%] Built target modules__fw_att_control
[ 94%] Built target modules__events
[ 94%] Built target modules__fw_pos_control_l1
[ 94%] Built target modules__gnd_att_control
[ 95%] Built target modules__gnd_pos_control
[ 96%] Built target lib__DriverFramework__framework
[ 97%] Built target boards_common
[ 97%] Built target board_stm32
[ 98%] Linking CXX executable ../../../nuttx_px4fmu-v2_default.elf
/home/abdularis/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: ../../../nuttx_px4fmu-v2_default.elf section `.data' will not fit in region `flash'
/home/abdularis/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: region `flash' overflowed by 221 bytes
collect2: error: ld returned 1 exit status
src/firmware/nuttx/CMakeFiles/nuttx_px4fmu-v2_default.elf.dir/build.make:195: recipe for target 'nuttx_px4fmu-v2_default.elf' failed
make[3]: *** [nuttx_px4fmu-v2_default.elf] Error 1
CMakeFiles/Makefile2:6542: recipe for target 'src/firmware/nuttx/CMakeFiles/nuttx_px4fmu-v2_default.elf.dir/all' failed
make[2]: *** [src/firmware/nuttx/CMakeFiles/nuttx_px4fmu-v2_default.elf.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:153: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 2

already try make distclean and re run make px4fmu-v2_default but still get the same error on 98%

aaumar, 2017-10-22:
please close it. fixed with ninja.
I would suggest the team to add install ninja build before make anything because on the dev site there is no instruction to install ninja build. Thank you dev team

closed 2017-10-22 by aaumar

1184, 2018-03-11:
I have the same problem. Did you manage to find the solution?

cdlwhm1217096231, 2018-03-30:
I have the same problem. Did you manage to find the solution?