PX4/PX4-Autopilot #8171: minor things to check before v1.7.0

dagar, 2017-10-22:
- cmake Makefile issues not present with the Ninja generator (check PX4 NuttX builds)
 - windows build
 - Nuttx stable branch naming
 - CI builds and metadata s3 deployment in sync with QGroundControl
 - check consistent tagging and version reporting (PX4 command line, QGC, flight review)

dagar, 2017-11-15:
https://github.com/PX4/Firmware/pull/8284

LorenzMeier, 2017-11-21:
Pixhawk mini flashing is standing - @bkueng could you please take a stab at that with @dagar?

dagar, 2017-11-21:
I'm not sure if this is new, but it's concerning. Recent snapdragon log from the test team had many instances of most topics.

https://logs.px4.io/plot_app?log=2943d850-0853-40f3-b129-63f6e4c1e29e
https://logs.px4.io/plot_app?log=b6a86822-c2d5-496c-9261-c757e3604992

[image]

LorenzMeier, 2017-11-24:
@MaEtUgR Could you comment on wether Windows builds right now? Is your toolchain ready to be released?

MaEtUgR, 2017-11-25:
@LorenzMeier I didn't try the "original" old windows toolchain but last time I checked (already pretty long time ago) the build was broken. The new Toolchain is not ready yet, see https://github.com/PX4/Firmware/issues/8357 where I'll also post progress. I would not make the current release dependent on that. Correct me if I'm wrong but in my eyes Windows support is not just recently a problem otherwise I would have used it before.

closed 2017-12-08 by dagar