PX4/PX4-Autopilot #8174: Possible to simulate motor failure with Pixhawk in Gazebo??

siddhu45, 2017-10-23:
Hello community,

I wanted to simulate a motor failure for a quad with Pixhawk in gazebo. Is it possible to do so? I wanted to try my own high-level controller with the low-level controller of Pixhawk. It might be possible to do so by actuator control but I don't want to do that. I wanted to check whether my high-level position controller can stabilize the motor failure or not.
Any help or guidance towards the relevant direction is appreciated.
Thanks.

PS: I am not an expert in Gazebo but have used it enough for a few months. So, be gentle on the technical language! Thanks.

siddhu45, 2017-10-24:
Hi! Can anyone please help.
Is it possible to write a plugin in gazebo which would stop the motion of  a link while in simulation when I command, that would also be great. Awaiting your help. Thanks.

dagar, 2017-10-25:
This is more of a general gazebo question. I could show you how to fake it from the firmware side, but it's probably better to handle it in gazebo. Please let me know if you find a solution as it might be helpful for general PX4 testing.

siddhu45, 2017-10-26:
Hi, thanks for replying. Actually I don't have a perfect solution but as a work around, I modify the `gazebo_motor_model` plugin and set a failure flag to the motor I want to fail in the sdf file. This actually works and makes the motor I want to fail, fail. But what I am trying to do is implement a subscriber in the plugin and change the failure flag online on my command.
Let me know if this was helpful, maybe I will update once I am able to subscribe to a node in the plugin and the failure works while in simulation.
Cheers!

closed 2017-10-26 by siddhu45

TSC21, 2017-10-26:
That seems useful. If you find it useful for others also, please open a PR to `sitl_gazebo`. Thanks!

siddhu45, 2017-10-26:
Sure will do.

siddhu45, 2017-11-01:
@TSC21 I am able to subscribe to a ROS topic in gazebo motor plugin to perform in flight motor failure simulation tests. If you think this is useful, I can create a PR after this weekend. Let me know.

TSC21, 2017-11-01:
> @TSC21 I am able to subscribe to a ROS topic in gazebo motor plugin to perform in flight motor failure simulation tests. If you think this is useful, I can create a PR after this weekend. Let me know.

Yes it is. Please, when you are able, launch a PR. I will review it then. Thanks!

S259420, 2019-01-21:
@siddhu45 
Hello, could you please explain how did you solve the problem? I'd like to do something similar...

siddhu45, 2019-01-30:
Hi @S259420 You can have a look at this [PR](https://github.com/PX4/sitl_gazebo/pull/141). You can look at further development in the same plugin in [sitl_gazebo](https://github.com/PX4/sitl_gazebo) repository. Hope this helps.
Cheers!

kang-hyeok, 2021-11-28:
Hello, I'd like to arbitrarily try a motor failure situation on another model of Gazebo I made. Currently, a px4 Gazebo environment is being built in ubuntu18.04. Can you help me?