PX4/PX4-Autopilot #8178: Incorrect data type for mount parameters MNT_OB_LOCK_MODE and MNT_OB_NORM_MODE

mdailey, 2017-10-24:
The parameter reference specifies that the two parameters MNT_OB_LOCK_MODE and MNT_OB_NORM_MODE are type float.

However, src/drivers/vmount/vmount.cpp has them declared as

```
	int32_t mnt_ob_lock_mode;
	int32_t mnt_ob_norm_mode;
```

This may not cause a problem for most users, as these parameters are only used in AUX output mode, and due to the parameter type system, user entered values of 0, 1, and -1 will end up getting range limited to PWM_AUX_MAX and PWM_AUX_MIN by the mixer as expected.

However, intermediate floating point values such as 0.2 or -0.2 get interpreted as int32 then range limited by the mixer, so instead of an expected PWM output, say, of 1600 for a MNT_OB_LOCK_MODE of 0,2, we might get 2000.

Changing the type declarations to float fixes the problem.

bkueng, 2017-10-24:
That's wrong indeed. Can you send a PR? And please take note of #8149 for potential conflicts (fyi @mje-nz)

mdailey, 2017-10-24:
OK, done! The type is correct everywhere else in vmount. My manual testing gives correct PWM output of 1600 and 1400 for +0.2 and -0.2 with default PWM_AUX parameters and the default mixer.

closed 2018-02-21 by PX4BuildBot