PX4/PX4-Autopilot #8184: A serial port bug!!!!!!!!!!!!!!!!!!!!!!!!!!!

yangang123, 2017-10-25:
Good evening:

   I wrote a very simple serial port driver, when I write data to the serial port, after more than 200 ms delay, NSH crash will occur!!!  

```
#include <fcntl.h>
#include <stdio.h>

extern "C" __EXPORT int serial_buggy_code_main(int argc, char *argv[]);

int serial_buggy_code_main(int argc, char *argv[]) {
	uint8_t buffer[16];
	int fd_ = ::open("/dev/ttyS1", O_RDWR | O_NOCTTY);
	for (int i=0; i<sizeof(buffer)/sizeof(uint8_t); i++) {
		buffer[i] = '0'+i;
	}

	int stime = atoi(argv[1]);
	if (fd_ > 0) {
		for (int j=0; j<2; j++) {
			int ret = ::write(fd_, buffer, 16*sizeof(uint8_t));
			printf("%d\n", j);
			if (ret < 0) {
				printf("write fail:%d", ret);
			}
			usleep(stime*1000);
		}
		printf("try close\n");
		::close(fd_);
		printf("closed\n");
	}

	return 0;
}
```

dagar, 2017-10-25:
Which board and configuration? Can you shared the console output when it crashes?
I'm wondering if the mavlink module is still active and trying to use that serial port simultaneously.

yangang123, 2017-10-26:
dagar： Thank you for your reply.

**when I write data to the serial port, after more than 200 ms delay, NSH crash will occur???**

**1.  The nsh console output**
```
$ screen /dev/ttyACM0

nsh>
nsh> ps
  PID PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK  COMMAND
    0   0 FIFO     Kthread --- Ready              00000000 Idle Task
    1 192 FIFO     Kthread --- Waiting  Signal    00000000 hpwork
    2  50 FIFO     Kthread --- Ready              00000000 lpwork
    3 100 FIFO     Task    --- Waiting  Semaphore 00000000 init
    5  70 FIFO     Task    --- Running            00000000 nshterm /dev/ttyACM0
nsh> ver all
HW arch: PX4FMU_V2
FW git-hash: f160743df40b9dcc9558f3df1c6eeaf56646a758
FW version: 1.6.5 0 (17171712)
OS: NuttX
OS version: Release 7.a.0 (118096127)
OS git-hash: 8b81cf5c7ece0c228eaaea3e9d8e667fc4d21a06
Build datetime: Oct 26 2017 13:31:16
Build uri: localhost
Toolchain: GNU GCC, 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
MFGUID: 383737303435511600370018
MCU: STM32F42x, rev. 3
UID: 370018:34355116:38373730
nsh>
nsh>
nsh> serial_buggy_code 100
0
1
try close
closed
nsh> serial_buggy_code  200
0
1
try close

```

**2.**  
```
HEAD detached at v1.6.5
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)

	modified:   ROMFS/px4fmu_common/init.d/rcS
	modified:   cmake/configs/nuttx_px4fmu-v2_default.cmake

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	src/drivers/serial_buggy_code/

no changes added to commit (use "git add" and/or "git commit -a")
```

**3. ROMFS/px4fmu_common/init.d/rcS**
```
#!nsh

mount -t procfs /proc

#
# Start CDC/ACM serial driver
#
if sercon
then
fi

nshterm /dev/ttyACM0 &
```

yangang123, 2017-11-14:
closed #8184

dagar, 2018-01-22:
Was this resolved?

yangang123, 2018-01-26:
Hi @ dager
_**I try to solve this problem, I find a USART2 enable the flow control, however, I just connect 2 line, TX and RX, and then I comment in the defconfig file to  USART2 flow control, then test the application again, found that everything was normal**_

**## Before the change:** 
```
#
# USART2 Configuration
#
CONFIG_USART2_RXBUFSIZE=600
CONFIG_USART2_TXBUFSIZE=1100
CONFIG_USART2_BAUD=57600
CONFIG_USART2_BITS=8
CONFIG_USART2_PARITY=0
CONFIG_USART2_2STOP=0
CONFIG_USART2_IFLOWCONTROL=y
CONFIG_USART2_OFLOWCONTROL=y
# CONFIG_USART2_DMA is not set
```

## **The current:** 
```
#
# USART2 Configuration
#
CONFIG_USART2_RXBUFSIZE=600
CONFIG_USART2_TXBUFSIZE=1100
CONFIG_USART2_BAUD=57600
CONFIG_USART2_BITS=8
CONFIG_USART2_PARITY=0
CONFIG_USART2_2STOP=0
#CONFIG_USART2_IFLOWCONTROL=y
#CONFIG_USART2_OFLOWCONTROL=y
# CONFIG_USART2_DMA is not set
```

closed 2018-08-06 by santiago3dr