PX4/PX4-Autopilot #8186: VTOL crash on V4Pro

santiago3dr, 2017-10-25:
vtol crash 24/October

crash log:
https://review.px4.io/plot_app?log=da864870-aa30-4133-be64-c766e2e0e9e7

previous successful flights on same platform:
https://review.px4.io/plot_app?log=1fa2997a-b9d8-4039-af2e-dad379be803c
https://review.px4.io/plot_app?log=deacbcaa-61ed-41c2-80d2-9b636b66a983

Flight conditions:
aprox 27°C degrees
moderate winds, 4~5 m/s

Sequence of events:
auto take-off in MC
failed auto-transition to FW
retook manual control in stabilized and successfully engaged manual transition to FW
engaged mission; vtol headed in correct direction to first waypoint

after a few waypoints, vtol started to veer off coarse and switched back to MC
switched MC to stabilized at which point it was hard to control and started to dip in pitch and roll violently while losing altitude

@dagar

dagar, 2017-10-25:
Multiple primary IMU timeouts.
[image]

[image]

Primary IMU (mpu6000 driver) reporting bad transfers.

[image]

dagar, 2017-10-25:
The selected sensor changed as a result of the timeout, but then reverted, timed out again, etc.

[image]

To start with we need to take a look at the sensor voting and switching logic.
@priseborough @LorenzMeier

priseborough, 2017-10-25:
I'm not familiar with that part of the code, but the requirement for handling timeouts and other hard faults with duplex inputs should be to switch if the current active sensor instance faults to the next instance that is not faulted. The switch should be sticky so that the new active instance is kept until it faults, in which case the selector goes to the next instance that is passing.

priseborough, 2017-10-25:
The estimator will need to be made aware of the switch so that it can reset the IMU bias states and associated covariances.

priseborough, 2017-10-25:
I will submit a PR adding the method to the ecl enabling the bias states and variances to be reset. If the sensor combined message can provide a uint8  indicating the source sensor index, then I can use that in ekf2_main.cpp to call the reset function.

priseborough, 2017-10-26:
Here's the supporting changes for ecl https://github.com/PX4/ecl/pull/348

When changes to the sensor module have been made, I'll put together the required ekf2_main changes.

bkueng, 2017-10-26:
@priseborough you can use the existing `sensor_selection` topic. And the failover logic is already in ecl: https://github.com/PX4/ecl/blob/master/validation/data_validator_group.cpp#L148

mrpollo, 2017-11-08:
any update on this one? are we good to close this?

priseborough, 2017-11-08:
It's resolved

closed 2017-11-08 by priseborough