PX4/PX4-Autopilot #8192: ecl lib geo

Stifael, 2017-10-26:
geo functionality is used in the firmware as well as in ecl. I am wondering if it is not worth the effort to combine them and make a separated lib or at least just use the one from ecl and whatever after closing the diff between ecl -geo and firmware-geo. I prefer to create a separated lib such that it can be used by other projects as well.

Stifael, 2017-10-26:
I think ecl requires a general clean up. L1 even uses mathlib::Vector which is not part of ecl and is also not a submodule either

dagar, 2017-10-26:
I was imagining geo as another top level ECL component. Each would be built as a static/shared/both lib. Firmware would include the directory and each PX4 module would link to the relevant component. EKF2 would link the ecl_ekf lib, FW position would link L1, etc. If a Firmware build didn't include that module (ekf2, fw_pos_ctrl_l1, fw_att_ctrl_l1) then the ecl lib would have no dependency and wouldn't be built.

https://github.com/PX4/ecl/issues/346

closed 2018-02-21 by PX4BuildBot