PX4/PX4-Autopilot #8196: Position Controller Mixing

imcnanie, 2017-10-26:
**Feature Request**

Is there a standard way to mix a tilt-quadcopter so attitude is controlled with rotor speed, and position is controlled by rotor tilt? If not, will implementing this required modifying mc_pos_control?

In other words, the copter body would remain flat using torque from rotor speed. It would translate using rotor tilt.

This feature would also be useful for a pusher quad.

bresch, 2017-10-31:
Hi @imcnanie , at the moment the tilt-rotor mechanism is handled in the VTOL tiltrotor code (https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/tiltrotor.cpp) but the tilting mechanism is only used to make a VTOL transition.
For a pusher quad, this is also handled for standard VTOLs. However, the implementation is not nice and I would not recommend you to do the same with the tiltrotor (the VTOL code takes the pitch setpoint, converts it to a pusher command et republish the pitch setpoint to zero. This is more a hack than a proper implementation).
The proper way to do it is in discussion since quite a lot of time with the other VTOL developers and would require several changes in the `mc_pos_control` code. If you have time and motivation you can start to think about a nice way to do it and we could work together. Otherwise, just hack the tiltrotor code, it should the the easiest way to make something rapidly functional.

imcnanie, 2017-11-17:
Hi @bresch ! I'm very interested in helping re-implement this. How can I be part of the discussion?

dagar, 2017-11-18:
@jlecoeur FYI

bresch, 2017-11-20:
Hi, @imcnanie , @jlecoeur is actively working on the control of a fully-actuated multicopter. His copter doesn't tilt to move but rather uses lateral thrust. (https://github.com/jlecoeur/Firmware/pull/1)
Using the modification he made in the Multicopter Position Controller to publish a 3D thrust and then writing a tiltrotor mixer that inherits from his mixer `mixer_multirotor_6dof`, you should be able to make your tilting mechanism moving the drone correctly.
I suggest that you check with him the feasibility of this approach.

closed 2018-02-21 by PX4BuildBot