PX4/PX4-Autopilot #8197: Travis is not returning current master commit

r0gelion, 2017-10-26:
Travis is returning commit 69231ea but there are 2 new commits [merged to master](https://github.com/PX4/Firmware/commits/master).

To reproduce you can load master from QGC or you can see what Travis is returning here:
http://px4-travis.s3.amazonaws.com/Firmware/master/px4fmu-v4_default.px4

r0gelion, 2017-10-26:
Now is returning the latest commit (4416c4d)

I don't know what changed but it's working now.

r0gelion, 2017-11-02:
Now is returning commit 29af234 but current master is 459a71a

dagar, 2017-11-02:
Thanks, could you share exactly where you're seeing these?

r0gelion, 2017-11-02:
Here:
http://px4-travis.s3.amazonaws.com/Firmware/master/px4fmu-v4pro_default.px4
http://px4-travis.s3.amazonaws.com/Firmware/master/px4fmu-v4_default.px4
http://px4-travis.s3.amazonaws.com/Firmware/master/px4fmu-v3_default.px4
http://px4-travis.s3.amazonaws.com/Firmware/master/px4fmu-v2_default.px4

dagar, 2017-11-02:
I believe it's actually a build system issue. If the build fails on master (something people don't directly watch) it might not update.

[image]

For now I'll keep an eye on it and make sure these master builds are properly getting deployed. Soon I'd like to migrate to our own CI system (http://ci.px4.io/blue/organizations/jenkins/) to fix a lot of these problems that are out of our control.

This is current master I've forced to rebuild. https://semaphoreci.com/px4/firmware/branches/master/builds/1878

dagar, 2017-12-06:
Caused by intermittent semaphoreci failure.

CI improvements (especially stability!) will be a focus after v1.7.0.

closed 2017-12-06 by dagar