PX4/PX4-Autopilot #8198: Battery remaining inversed when compensating for load

sanderux, 2017-10-26:
The battery remaining function is broken on today's master.
It increases drastically when under load (if anything it should decrease). Tested with and without setting internal resistance, both behave the same. The voltage reading does read as expected.

[image]

Log from other flight same behavior:
https://logs.px4.io/plot_app?log=7f323da0-8f75-46f1-aba9-f0a68150b691

ping @LorenzMeier @dagar

sanderux, 2017-10-26:
ping @davids5

dagar, 2017-10-27:
For VTOL this would make sense if you're relying on the normalized throttle usage (https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/battery.cpp#L179), but I see you have BAT_R_INTERNAL set.

LorenzMeier, 2017-10-30:
@sanderux This is still persisting, right?

sanderux, 2017-10-31:
yes

sanderux, 2017-10-31:
@dagar it happens with with and without r_internal

MaEtUgR, 2017-10-31:
@sanderux I'm currently revising the battery estimation. It's broken in several ways according to my opinion. Not sure if your case is covered in my fixes, let's investigeate.

On the first glance it looks to me as if your battery estimation is always corrected using the thrust and for the factor of how much this is done you can use the parameter `BAT_V_LOAD_DROP`. You even say it does not change when setting an internal resistance.

What you should also know is that the current estimation always adjusts the empty voltage compared to the measurement itself which results in rescaling the whole calculation... If you feel like and have time you can test my pr as soon as I fixed the CI stuff and pushed the newest enhancements today.

sanderux, 2017-11-03:
I re-checked all the math and found that it was indeed related to internal resistance that was likely wrong for this lipo. Initially i thought it was not related as it exhibited the same behavior when bat_IR was disabled. but on the bench test it did not (perhaps param didnt save in field test).

Closing for now.

closed 2017-11-03 by sanderux

nskjoung, 2017-11-15:
@sanderux i'm be in a same situation. how does it clear?

MaEtUgR, 2017-11-15:
@nskjoung I think his conclusion was that the internal resistance parameter was not set correctly and hence the measurement got overcompensated. If you want to try supposedly better battery estimation please try #8153 . It will likely just not be in 1.7 because it requires a lot of testing.