PX4/PX4-Autopilot #8204: position lock

alirezaka, 2017-10-28:
I tried to follow <<Launching Gazebo with ROS wrappers>> in here https://dev.px4.io/en/simulation/ros_interface.html but when I use << commander takeoff>> the error << rejecting takeoff, no position lock yet. Please retry.. >> appears and I don't the reason.
could someone give me some hints about it ?

LorenzMeier, 2017-10-28:
Wait until the system indicates position lock and try again.

closed 2017-10-28 by LorenzMeier

alirezaka, 2017-10-28:
thanks for your attention but I wait for such a long time but nothing
appears

alirezaka, 2017-10-28:
just this ->
INFO  [simulator] Got initial simulation data, running sim..
INFO  [pwm_out_sim] MODE_8PWM
INFO  [tone_alarm] startup
ERROR [land_detector] Timed out while waiting for thread to start
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14556 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14557 remote port 14540
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [logger] pxh> logger started (mode=all)

zubair123khan, 2018-05-30:
i am facing the same thing, i have my mavros running and when do commander takeoff, its showing no position lock yet

alirezaka, 2018-05-31:
The problem is your GPS connectivity , check all aspects of it.

zubair123khan, 2018-05-31:
I checked all, I am not able to use commander takeoff. Moreover I am now using launch file from mavros posix_sitl.launch and I tried using takeoff and arming from rossservice and I was unable to do that, everytime I am setting my arm to true using rosservice , it's returning me false.

alirezaka, 2018-06-01:
I myself couldn't figure out how to solve this problem, so I gave it up and instead I tried to use MavLink outside ROS environment.
That's all help I can give you.