PX4/PX4-Autopilot #8205: ROS

alirezaka, 2017-10-28:
I did catkin_make on Firmware folder in my computer and it finished cleanly and without any error but when I
did << roslaunch px4 posix_sitl.launch >> following error appeared but I don't know why ?
Invalid <arg> tag: mavlink_sitl_gazebo
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/alireza/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share. 

Arg xml is <arg default="$(find mavlink_sitl_gazebo)/worlds/empty.world" name="world"/>
The traceback for the exception was written to the log file.

could someone please tell me the reason ?

TobiMiller, 2017-11-13:
you need to source the firmware folder

alirezaka, 2017-11-13:
Hi Tobias
you mean :

source Tools/setup_gazebo.bash $(pwd)
$(pwd)/build/posix_sitl_defaultexport
ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)export
ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
?
I have sourced it but still my ROS master doesn't find it !

TobiMiller, 2017-11-15:
but you were in your firmware folder while sourcing it?

alirezaka, 2017-11-15:
I created a catkin_ws and I included firmware in it , I did catkin_make and it finished cleanly without any error but when  lauched 《 roslaunch px4 posix_sitl.launch 》the error I mentioned aboved appeared.

alirezaka, 2017-11-15:
Should I be in firmware folder ?

TobiMiller, 2017-11-16:
`
cd %FIRMWARE PATH%
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/posix_sitl_defaultexport ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo`

alirezaka, 2017-11-16:
Thanks Tobias

closed 2018-02-21 by PX4BuildBot