PX4/PX4-Autopilot #8213: Tailsitter mixer is broken (in gazebo)

imcnanie, 2017-10-30:
**Bug Report**

Hello. The tailsitter model appears to be broken. Can anyone reproduce it for me? I'm running PX4 Master.

Take off works great, no issue there. Transition appears functional too. But in plane mode the vehicle quickly becomes unstable.

dagar, 2017-10-30:
I'm aware of this, but haven't had time to fix it. Can you share a log? https://logs.px4.io/
After the v1.7.0 release there's going to be a big push for significantly more automated testing. It would be great to get this working again with testing so we can keep it working.

imcnanie, 2017-10-31:
Here you go: https://logs.px4.io/plot_app?log=cf794d8f-59b5-476b-8143-00612c6d7ba4

[image]

What do you think the issue is. I messed around with the mixer for a while, it didn't seem like roll control was working, and the pitch "virtual elevator" was reversed. 

Also, why do we have a virtual elevator?

tumbili, 2017-10-31:
Hi @imcnanie 
I gave the tailsitter config a try in master and could not even do a takeoff, because it would complain of not having a position lock yet. If you have time can you try the following steps:

1) Use this mixer
```
Mixer for Tailsitter with x motor configuration and elevons
===========================================================

This file defines a single mixer for tailsitter with motors in X configuration.  All controls
are mixed 100%.

R: 4x 10000 10000 10000 0

Z:

# left elevon
M: 1
O:      10000  10000      0 -10000  10000
S: 1 0  10000  10000      0 -10000  10000

# right elevon
M: 1
O:       10000   10000    0 -10000  10000
S: 1 0   -10000   -10000    0 -10000  10000

# mixer for the virtual elevator
M: 1
O:       10000   10000   0 -10000  10000
S: 1 1   10000   10000    0 -10000  10000

```

2) Add this line to the elevator plugin here
https://github.com/PX4/sitl_gazebo/blob/b052c97f7c3a2c39ab8ec06ae79c66431f16c659/models/tailsitter/tailsitter.sdf#L532

```
<control_joint_rad_to_cl>-4.0</control_joint_rad_to_cl>
```

Regarding your question about why the virtual elevator is needed:
The liftdrag plugin does not model any aerodynamic moments (Cm = 0 all the time).
So you cannot use elevons on the wings to create a pitching moments. That's why we introduced a virtual elevator which takes care of that.

Let me know how it goes, it would be nice to have at least a functional model of the tailsitter.

imcnanie, 2017-10-31:
After that, it is still uncontrollable. For me I get the position lock issue with all the SITL models, I just have to wait a bit until it gets a home position.

Are you saying that the gazebo <cma>0.0</cma> tags don't work? That's disappointing. After working with the sim for a while, I can think of a few ways to make it better. Is there a set of devs working on it?

imcnanie, 2017-10-31:
Haha check out the gazebo liftdrag plugin! https://github.com/arpg/Gazebo/blob/master/plugins/LiftDragPlugin.cc#L276

tumbili, 2017-10-31:
@imcnanie That's what I told you ;-)

closed 2018-02-21 by PX4BuildBot

ADITYA97-CODER, 2024-12-08:
@imcnanie I aim still facing this issue. Have you found the solution. Also I couldn't find any mixer files in my repo.

Jaeyoung-Lim, 2024-12-08:
@ADITYA97-CODER It is not possible the encounter the same issue, as mixers have been deprecated.

Please post a new issue that describes your problem