PX4/PX4-Autopilot #8222: Sensiron pitot pressure loss correction appears to be incorrect

priseborough, 2017-11-01:
The calculation of differential pressure drop due to the pitot tube for a given flow rate asumes a fixed air density: 

https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/airspeed.c#L91

This appears to be inconsistent with the use of density when calculating the flow rate through the sensor and pressure drop in the tubing which  indicates that a volumetric, not mass flow rate is being used:

https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/airspeed.c#L82
https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/airspeed.c#L100

Additionally, the sensor tube length parameter does not specify if it is the sum of pitot and static tube length or tube length from sensor to pitot tube, and does not specify the assumed tube diameter.

acfloria, 2017-11-01:
Sensirion already provided a new model, you can find it [here](https://github.com/PX4/Firmware/issues/8140).

I adopted the code and already did a flight test. I will do another one this friday and if it looks good I will merge the new model.

dagar, 2017-12-06:
I believe this was resolved by @acfloria in https://github.com/PX4/Firmware/pull/8242. Please reopen if I'm mistaken.

closed 2017-12-06 by dagar