PX4/PX4-Autopilot #8224: Logging custom topics in SITL with gazebo

ningyuanz, 2017-11-01:
Hello,

I have been trying to log a custom topic in simulation using gazebo. I have successfully logged the topic in real flights with the pixhawk hardware by changing the code of sdlog2. However, it looks like the SITL is using the new logger to generate ulog files and I failed to log the data I want. Below are the things I tried:

1. Setting SYS_LOGGER to 0 to use the old logger. However I could not find the parameter in QGC and where to change the parameter in source codes.

2. Stop logger when the simulation is running through the shell and the start sdlog2 manually. But the log file I got from QGC is still in the format of .ulg and it is somehow broken that I cannot use ulog2csv to decode it.

3. Tried to change the source codes of 'logger' but could not figure out how it works.

Could anyone give me some suggestions? Anything will be appreciated.

bkueng, 2017-11-01:
Hi @ningyuanz. With the new logger you can create a text file with your topics: https://dev.px4.io/en/log/logging.html#configuration. For SITL, it's under `build/posix_sitl_default/tmp/rootfs/fs/microsd/etc/logging/`.
The list of default topics is here: https://github.com/PX4/Firmware/blob/master/src/modules/logger/logger.cpp#L572

ningyuanz, 2017-11-04:
Thank you very much @bkueng. Problem solved.

closed 2017-11-04 by ningyuanz