PX4/PX4-Autopilot #8225: 1.7 Release notes

mrpollo, 2017-11-01:
Let's start adding the release notes for 1.7 here

tops4u, 2017-11-03:
Added Support for Lightware SF20/LW20 I2C connected Lidars.

Ahrovan, 2017-11-12:
Please provide changes and development details for version 1.7

AlexisTM, 2017-11-15:
Solved bugs for the ECL library (EKF2) which impacted the PX4Flow #8102

dagar, 2017-11-15:
@jlecoeur do you want to add something high level about the mixer updates? Migration for users that had custom mixers, etc.

jlecoeur, 2017-11-15:
Auto-generate multirotor mixers from files that describe their geometry. See here <link to dev guide page to be done https://github.com/PX4/Devguide/issues/349 > how to migrate or add custom mixers.

scott-eddy, 2017-11-15:
FMU can now be run as a task #7640

mrpollo, 2017-11-15:
Preliminary changelog https://gist.github.com/mrpollo/0b4e8985e829de1961a47909376a2b85

mrpollo, 2017-12-20:
https://github.com/PX4/Firmware/releases/tag/v1.7.0

closed 2017-12-20 by mrpollo