PX4/PX4-Autopilot #8227: "No Local Position" error on takeoff with plane (SITL Gazebo) (log attached)

mikeklinker129, 2017-11-02:
I am flying the master branch, however I have seen the same behaviour with release v1.6.5.
[Link to log file](https://logs.px4.io/plot_app?log=8d2c8589-1951-4764-bc8f-1e023d8e01ca)

I am seeing a variety of failures shortly after take off, starting with a failsafe due to 'no local position'. 

Here is my process (running ubuntu 16.04):
1. Set environment variables to define the home position for the SITL sim. 
```
export PX4_HOME_LAT=42.0
export PX4_HOME_LON=-71.0
export PX4_HOME_ALT=98.0
```
2. Start gazebo_plane simulator 
```
make posix gazebo_plane
```
3. Connect QGCS to monitor
4. Within the PX4 terminal, command it to takeoff
```
pxh> commander takeoff
```
The aircraft starts to take off, but shortly after lift off it triggers a failsafe event due to no local position. 

Terminal readout: 
```
INFO  [logger] logger started (mode=all)
pxh> INFO  [logger] Start file log
INFO  [logger] Opened log file: rootfs/fs/microsd/log/2017-11-02/18_11_46.ulg
[Wrn] [msgs.cc:1655] Conversion of sensor type[imu] not suppported.
INFO  [mavlink] partner IP: 127.0.0.1
[Wrn] [msgs.cc:1655] Conversion of sensor type[imu] not suppported.
INFO  [lib__ecl] EKF aligned, (pressure height, IMU buf: 15, OBS buf: 14)
[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/plane/motor_speed/4, deleting message. This warning is printed only once.
INFO  [lib__ecl] EKF GPS checks passed (WGS-84 origin set)
INFO  [lib__ecl] EKF commencing GPS fusion
INFO  [commander] home: 41.9999999, -71.0000001, 98.11
INFO  [tone_alarm] home_set

pxh> 
pxh> commander takeoff
pxh> INFO  [tone_alarm] positive
INFO  [commander] home: 41.9999998, -71.0000000, 97.92
INFO  [tone_alarm] arming
INFO  [commander] Takeoff detected
WARN  [commander_tests] Failsafe enabled: no local position
INFO  [tone_alarm] fast_bat
WARN  [navigator] Global position failure: fixed bank loiter
WARN  [lib__ecl] EKF GPS fusion timeout - reset to GPS
INFO  [commander] Landing detected
WARN  [lib__ecl] EKF GPS fusion timeout - reset to GPS
```

I do not see this behavior when I fly the simulator in the stock home location (Switzerland). I have tried a few locations in the USA, and they have demonstrated a similar failure. 

I have tried flying my plane with a pixracer running the same software in Massachusetts (close to where I am starting the simulator), and I have seen similar error messages, occurring right after takeoff: 
```
0:01:10 WARNING: [commander_tests] NOT ARMING: Press safety switch first!
0:01:10 WARNING: [commander] Arming not possible in this state
0:04:01 WARNING: [lib__ecl] EKF bad yaw corrected using GPS course
0:04:21 WARNING: [lib__ecl] EKF GPS fusion timeout - reset to GPS
0:04:28 WARNING: [lib__ecl] EKF GPS fusion timeout - reset to GPS
0:04:35 WARNING: [lib__ecl] EKF bad yaw corrected using GPS course
0:04:35 WARNING: [lib__ecl] EKF GPS fusion timeout - reset to GPS
```
The yaw errors are leading me to believe that there is some magnetometer problem, however I am not sure. Another interesting note is that when I start the gazebo simulator in Switzerland (stock location), GQCS indicates that the plane is pointed directly East, at 90deg. When I start the simulation in MA, USA, gazebo still indicates that the plane should be pointed East, however QGCS shows a heading of ~58 degrees. I tried setting the EKF declination to -14.5, which is the declination in my area, however it seemed to have no effect on this startup discrepancy and the take off failsafe behaviour.

anassinator, 2017-11-02:
I believe this is a duplicate of #6843
This seems to fix it for me: PX4/sitl_gazebo#96

mikeklinker129, 2017-11-02:
Thanks for the heads up, I will try that fix and report back.

mikeklinker129, 2017-11-03:
Confirmed that that declination reverse fixed it.

closed 2017-11-03 by mikeklinker129