PX4/PX4-Autopilot #8228: Add MAV_CMD_CONDITION_YAW

LorenzMeier, 2017-11-02:
We need this for QGC structure scan.

http://mavlink.org/messages/common#MAV_CMD_CONDITION_YAW

DonLakeFlyer, 2017-11-02:
Also Mavlink spec is not specific enough. Should be updated for this:
Mission Param #2 - speed during yaw change:[deg per second], 0 - use default turn rate

DonLakeFlyer, 2017-11-02:
Hmm, now that I have this coded on the QGC side I realize this isn't quite what I want. Normally the vehicle points forward along the flight path (param controlled). But what I want is instead for it to point 90 degrees clockwise to the flight path as it flies the mission. Not quite sure how to do that.

DonLakeFlyer, 2017-11-02:
Example:
[image]

 Flies this:
[image]

With the non gimbaled vehicle always pointing at the structure to keep camera in right direction.

dagar, 2017-11-02:
What's the interpretation of these CONDITION commands?
Based on some of the others it sounds like MAV_CMD_CONDITION_YAW should block mission acceptance (advancement) until that yaw is achieved, but it sounds like you want to use it to command a yaw during mission?

DonLakeFlyer, 2017-11-02:
They do both, change to the yaw but they don't move to the next command until the yaw requested is reached. It need something which is yaw 90 degrees to the mission flight path though which this doesn't do.

DonLakeFlyer, 2017-11-02:
It's only sort of an ROI. Since the flight path really determines it, just with a 90 degreee offset. A paramater won't work since you don't want to futz a param just to get a misison to work.

dagar, 2017-11-02:
Yes I agree we shouldn't do it with a parameter. I meant adding a new ROI type for a particular yaw. Telling the vehicle higher level what you want (look at yaw 90 degrees) rather than specific moves to do it. Then based on the particular vehicle and configuration the firmware can do it either with the gimbal or yawing during the mission.

DonLakeFlyer, 2017-11-02:
So it would be something like:
MAV_RIO_WPNEXT_YAW_OFFSET - Point toward next waypoint with offset in yaw
And then you stick Yaw degress in param 4 or something.

DonLakeFlyer, 2017-11-02:
So if I used 90 degress with MAV_ROI_WPNEXT_YAW_OFFSET and flew a clockwise pattern it would always point to the structure.

dagar, 2017-11-03:
That sounds good to me, but probably needs a bit more thought overall.

closed 2018-02-21 by PX4BuildBot