PX4/PX4-Autopilot #8231: Hardware compatibility falied

anandbaskaran, 2017-11-03:
We are using PX4 for our drones with [this](https://hobbyking.com/en_us/hkpilot32-autonomous-vehicle-32bit-control-set-w-power-module.html?___store=en_us) hardware.Everything works like a charm with version 1.6.5.  But the board stops working if I upload the firmware from the current master branch.

Is this a bug or PX4 stopped supporting this specific hardware? It will be great if the release 1.7 ensures hardware compatibility!

anandbaskaran, 2017-11-07:
Here is a bit more insight (Thank you @jlecoeur for helping me)!

The master branch continues to work on this [specific hardware](https://hobbyking.com/en_us/hkpilot32-autonomous-vehicle-32bit-control-set-w-power-module.html?___store=en_us) just by reverting the commit 553c64925292d0fc99b46d9514eb6646472033e7. 

After this being done, when I issue the command `fmu sensor_reset 500` the entire systems reboots again. This signifies hardware limitation. I hope this issue will be considered for the upcoming release!

mhkabir, 2017-11-07:
@davids5

davids5, 2017-11-07:
The hardware that you're using is not FMUV2 compatible ( it may be an assembly issue). The  sensor power supply reset is wired incorrectly. When the sensors are reset it's actually powering down the MCU.

The manufacturer has checked recent production versions and the problem is not present. I would suggest you contact them and raise the issue.

closed 2018-02-21 by PX4BuildBot