PX4/PX4-Autopilot #8235: F450 with Pixhawk mini flyaway (vehicle recovered)

r0gelion, 2017-11-05:
Testing master with a DJI F450 frame with the Pixhawk mini took off and it started to fly away and didn't respond to RC inputs.
https://logs.px4.io/plot_app?log=d150bc14-7532-4476-9318-186df60a10b7

The unit crashed but was recovered.

bkueng, 2017-11-06:
Looks like a problem with estimation, but I also see jumps in the position setpoint. RC is lost only later.
@priseborough could you have a look?

priseborough, 2017-11-07:
The IMU is clipping due to high vibration. See rapid simultaneous rise in both vertical GPS velocity and baro vertical position innovations coincides with increase in Z accel vibration and throttle setting.

[image]

There is no evidence of estimator time slips.

closed 2018-02-21 by PX4BuildBot