PX4/PX4-Autopilot #8246: F450 with Pixracer didn't follow the mission, started climbing.

r0gelion, 2017-11-06:
Took off manually, good hover in position mode launched the mission and the unit started getting altitude but did not follow the mission.

https://logs.px4.io/plot_app?log=907d2032-92a3-43c8-9a71-1c18b0792194

korigod, 2017-11-08:
Setpoint triplets were empty, it looks like there was no mission onboard. Did you noticed `[commander] REJECT AUTO MISSION` warning? Vehicle never launched the mission, it switched to AUTO.LOITER instead.

SayanR1, 2017-11-08:
The vehicle did have a Mission on board, after landing the unit I double checked in QGC and ti was there.

dagar, 2017-11-17:
I'm thinking this might be more of a usability issue? There are many cases where it's not clear if the mission shown in QGC is synced and accepted by the vehicle.

r0gelion, 2017-12-14:
I'm closing this issue, if this happens again, I will re-open it.

closed 2017-12-14 by r0gelion