PX4/PX4-Autopilot #8247: Camera hot shoe feedback logging

JonReacher, 2017-11-07:
It seems that all or most of the pieces are in place to implement logging of camera feedback in place of the initial trigger to allow for higher precision in image location geotagging. Opening issue to initiate a public discussion.

 In order to avoid conflict with QGC geotagging as it stands, it seems that either the initial trigger or the feedback would be logged, but not both. This would probably mean a parameter for selecting which method to use.

dagar, 2017-11-07:
TODO
 - add this to the camera_feedback module - https://github.com/PX4/Firmware/tree/master/src/modules/camera_feedback
 - coordinate GPIO configuration with camera_trigger and px4fmu
 - review overall logging
  - should we log both the initial camera command AND the feedback?

JonReacher, 2017-11-07:
I would think that if it's possible to use the same message, but simply move it from trigger to feedback by selecting a parameter, that 1) QGC geotagging would be unaffected either way and 2) live on screen camera trigger indication wouldn't need any change to work, and in fact would work better by only showing successful triggers and indicating if a survey is not going well before downloading the images after.

 I.E., if the camera was turned off the current system would show camera triggers on screen, but this way would show that no successful triggers have actually happened.

mhkabir, 2017-11-17:
Small table indicating the use-cases I would like to support : 

|   | Hardware triggering | Hardware feedback | Geotagging needed * | `camera_feedback` | `camera_trigger` | Current status |
| :---: | :---: | :---: |  :---: | :---: | :---: | :---: |
| VIO (active triggering) | :heavy_check_mark: | :heavy_multiplication_x:  | :heavy_multiplication_x: | :heavy_multiplication_x: | :heavy_check_mark: | Works
| VIO (free-running) | :heavy_multiplication_x: | :heavy_check_mark: | :heavy_multiplication_x: | :heavy_check_mark: | :heavy_multiplication_x:| Implementing now
| VIO (closed-loop capture) | :heavy_check_mark: | :heavy_check_mark: | :heavy_multiplication_x: | :heavy_check_mark: | :heavy_check_mark: | Implementing now
| Photogrammetry (active triggering) | :heavy_check_mark: | :heavy_multiplication_x:  | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | Works
| Photogrammetry (free-running) | :heavy_multiplication_x: | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :heavy_multiplication_x:| Implementing now
| Photogrammetry (closed-loop capture) | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | Implementing now

*If geotagging is not needed, we send `CAMERA_TRIGGER` over mavlink. If geotagging is needed, we send `CAMERA_IMAGE_CAPTURED`. In no case do we need to send both, so we can optimize bandwidth use by proper selection.]

Also, 

> Should we log both the initial camera command AND the feedback?

Yes. I will add functionality in the driver to match up commands and feedback. We will still log 1 message for every commanded frame, and if feedback mode is enabled and a feedback signal recieved, the `camera_capture.result` field will be updated to 1 (commanded shot was successful). Otherwise it stays at 0 (i.e commanded but not captured). If feedback capture is not enabled (current state on master), it will stay at -1, indicating that no information about actual capture is available.

When this is complete, we can make QGC display the successful shots in green (result = 1), failed ones in red (result = 0) and no-feedback ones (result = -1) in blue.

@dagar @JonReacher How does that sound?

dagar, 2017-11-17:
I think that sounds good, but I need to understand a few more of the pieces in detail including how we coordinate configuration with px4fmu.

mhkabir, 2017-11-17:
So I'm currently pulling everything related to hotshoe capture out of px4fmu and into camera_feedback.

We coordinate with the FMU app in the exactly same way as we do for camera trigger. For trigger, rcS sets the FMU to pwm4, and uses the other 2 pins for trigger control. With feedback enabled, it will set FMU to pwm3. Great way, FMU will not interfere/reconfigure any pins it doesn't control.

dagar, 2017-11-18:
That sounds great for now. What I'm unsure about longer term is juggling the ownership of pins via rcS. It starts to fall apart when other modules want GPIOs, not to mention other boards.

JonReacher, 2017-11-19:
> When this is complete, we can make QGC display the successful shots in green (result = 1), failed ones in red (result = 0) and no-feedback ones (result = -1) in blue.

Code aside, the features aspect looks great.

closed 2018-02-21 by PX4BuildBot

cglusky, 2018-03-09:
@mhkabir I tried to follow the dev path here.  I think it's..

https://github.com/PX4/Firmware/pull/7201

and here

https://github.com/mavlink/qgroundcontrol/pull/5133

But it's not clear if there was support for shutter event feedback in any of that?

I suspect not since I do not see any docs on how to wire up the cable from the hot shoe.  Or google is failing me.

LorenzMeier, 2018-07-23:
@bresch Could you provide a timing update here?