PX4/PX4-Autopilot #8248: How yaw control in fixed-wing mode works

ningyuanz, 2017-11-07:
Hello everyone,

I am investigating the yaw control in fixed-wing mode by using SITL with gazebo to simulate a standard VTOL vehicle:

`make posix_sitl_default gazebo_standard_vtol`

My question is, is the yaw control the same as roll and pitch in fixed-wing? I saw in the `fw_att_control_main.cpp` that yaw control has parameters Ki, Kp and Kff but when I printed `control_input.yaw_rate_setpoint`, this value is always 0. However, the inner loop control input, `yaw_u`, is not zero. Does this mean the yaw control only has a PI + FF body rate controller? Or maybe it's because the VTOL model does't have a rudder?

Thanks.

dagar, 2017-11-07:
The yaw rate setpoint is calculated to maintain coordinated turns.

bresch, 2017-11-09:
Some facts to add to @dagar ' comment:
- The rudder is manly used for yaw damping
- Heading control with the rudder is only used during landing approach (https://github.com/PX4/Firmware/blob/cf7140526a41ad7fd209189a0d9bd33cf4bd0319/src/modules/fw_pos_control_l1/FixedwingPositionControl.cpp#L958)

ningyuanz, 2017-11-15:
Thanks for both comments. Really helps me understand the controls in FW mode.

closed 2017-11-15 by ningyuanz