PX4/PX4-Autopilot #8249: [Feature Request] How to control servo with respect to motor speed

hujin9190, 2017-11-07:
Hello!

I am trying to control a servo rotating automatically by reading the angular velocity data from brushless motor. The default setting is using RC to control the servo. Do you have any ideas on how to realize that?

Thank you!

dagar, 2017-11-07:
Hi @hujin9190,
Where are you getting the angular velocity data? The easiest way to control a servo is to add it to a mixer (https://dev.px4.io/en/concept/mixing.html), and then publish the appropriate actuator_control message.

hujin9190, 2017-11-07:
Hello, @dagar 

Thank you so much!

The angular velocity data is based on the main out PWM. Actually, I have no idea how to get the main out PWM. Is it possible to get it directly from the pwm.cpp sent_out_pwm() function?

Meanwhile, also stuck in the part that adds servo in the mixer. Can you give a detailed instruction on how to change the code in mixer part and publish the message? 

Thank you again!

korigod, 2017-11-08:
Are you sure your ESCs have linear pulsewidth vs RPM characteristic? As far as I know, all hobby grade ESCs don't.

closed 2018-02-21 by PX4BuildBot