PX4/PX4-Autopilot #8256: EKF2 Altitude estimation problem on Release 1.6.5 Stable

m91moreira, 2017-11-08:
Hi,

I'm experiencing a strange behaviour regarding the altitude estimation. 

The flight log is from a mission with 4 waypoints at 20 m height.

During the flight, I noticed some oscillations on the drone altitude, which I was able to confirm using an RTK GPS as ground truth ([https://imgur.com/DR6G0jY](Z)) .

On the flight log it's possible to see that both the GPS and barometer notice the altitude variation, but the estimated altitude only has a small change.

https://logs.px4.io/plot_app?log=1d070bb2-b001-48f1-b33c-73e347a17001#Nav-Altitude-Estimate

Did anyone experience this problem?

Thank you.

AlexisTM, 2017-11-13:
Hmmmm...

Looking at the log my guess is lack of power. I would propose to use a battery with more cells. Indeed; when moving along X and Y, the thrust axis is no more aligned with the gravity; thus you have to increase the thrust to keep the altitude. Here, the thrust is almost maximal all time. 

The other solution would be to reduce the payload of your copter.

m91moreira, 2017-11-14:
Hi AlexisTM,

Yes, the drone is a little bit heavy, but the fact that the motors are almost on the limit shouldn't be an input for the estimated altitude, right? If the barometer and the GPS notice the altitude change, the estimated altitude should also change.

AlexisTM, 2017-11-14:
Your copter is dropping. Then what is the cause, estimation is wrong, RTK is wrong, barometer is wrong, the copter is unable to achieve the mission? 

Reduce the weight/add a cell, fly again, take a video of the flight and post the log.

closed 2017-12-06 by dagar