PX4/PX4-Autopilot #8257: RTL issues with multiCopter

santiago3dr, 2017-11-08:
Quadcopter running pixracer (v4)
two possibly related issues
- upon engaging RTL, copter does not climb to set RTL altitude
- if engaging RTL twice in the same flight, the second instance of RTL veers in the wrong direction; heading is corrected half way through the return path

both issues present:
https://logs.px4.io/plot_app?log=a229d06a-b4f6-4333-ae1c-246c3f90f8b1
https://logs.px4.io/plot_app?log=c6d5b63b-c467-491b-83b3-cfb39064abeb

altitude issue present:
https://logs.px4.io/plot_app?log=f999958f-6a2f-419a-8a1c-e812045e8be7

using the same version of master on fixed wing, RTL looks to be working fine:
RTL set to 60m:
engaging rtl below set altitude:
https://logs.px4.io/plot_app?log=ac6e62a9-6c02-4e3d-a9c3-132c2adfd0c8

engaging rtl above set altitude:
https://logs.px4.io/plot_app?log=2ead01d6-a13d-4df6-b261-54c1ec2137ab

@dagar @sanderux @priseborough

dagar, 2017-11-09:
The veering in the wrong direction particularly concerns me.

dagar, 2017-11-09:
The climb is broken because the default MC max return altitude (-1 to disable) is respected. https://github.com/PX4/Firmware/blob/master/src/modules/navigator/rtl.cpp#L78

[image]

dagar, 2017-11-09:
@santiago3dr could you expand on the second point? Is it only the heading that's wrong, or does RTL actually fly in the wrong direction?

[image]

[image]

santiago3dr, 2017-11-09:
just ran 2 flights on the same copter to replicate this issue
https://logs.px4.io/plot_app?log=ddbab0b2-d1e5-4f3b-8371-7d30915499ba
https://logs.px4.io/plot_app?log=6cced4c1-6a59-4a23-a67f-68c72032ccea

none of the rtl attempts rose to set rtl altitude
in the first flight, the second and third rtl engagements veered in the wrong direction, correcting half way and on the second flight, this happened only on the third attempt at engaging rtl

santiago3dr, 2017-11-10:
was able to reproduce this on the snapdragon flight controller (EAGLE)
https://logs.px4.io/plot_app?log=5f0f6037-8fd8-416d-a0c4-f00b6494e677

it seems rtl finds a direct path to the first rtl trajectory
@dagar

MaEtUgR, 2017-11-12:
Could very well be related to https://github.com/PX4/Firmware/pull/6679#issuecomment-282674945
FYI @Stifael

dagar, 2017-11-12:
Fix for the altitude here - https://github.com/PX4/Firmware/pull/8272

Stifael, 2017-11-12:
Or we would bring this https://github.com/PX4/Firmware/pull/8101 in. 
That get_rtl_altitude function is not needed anymore

Stifael, 2017-11-13:
The problem with the strange rtl direction behavior sounds familiar, and the issue was caused by having the previous setpoint not reset correctly. I will look into it today.

dagar, 2017-11-14:
Please give this PR a try. https://github.com/PX4/Firmware/pull/8272

Stifael, 2017-11-14:
I confirmed now that one issue (there might be more than that) is that the previous setpoint get reset to the wrong setpoint. 
For instance:
- vehicle is in mission
- user/system demands rtl -> current setpoint is equal to home setpoint
- user demands again mission: previous setpoint gets set to current setpoint, which is set to home. Consequently, the vehilce will follow the line home-current setpoint.

dagar, 2017-11-14:
@santiago3dr various navigator fixes have made it into master. Please reopen if this isn't completely resolved.

closed 2017-11-14 by dagar

santiago3dr, 2017-11-14:
@dagar 
tested with master, 3 attempts at RTL, all successful; all rose to set altitude and took a direct path to home location
https://logs.px4.io/plot_app?log=4f166b9d-fd36-452c-be7b-9d2b600f1561

commit 668d2f3f (2017-11-14): RTL check alt_max before using

 - fixes #8257
commit 747dcc72 (2017-11-14): RTL check alt_max before using

 - fixes #8257
commit 9c3c557e (2017-11-14): RTL check alt_max before using

 - fixes #8257
commit 98cbd445 (2017-11-14): RTL check alt_max before using

 - fixes #8257