PX4/PX4-Autopilot #8259: [lpe] vision position fault problem

mandu46, 2017-11-09:
Hi. all
I have similar problem with [https://github.com/PX4/Firmware/issues/5714](url). 
In firmware 1.6.3 and lpe_fusion paramter set to 12 which is just fuse vision position and vision yaw, but sometimes [lpe] vision position fault message occurs and z value of "mavros/local_position/pose" is way off. It looks like this is because the PX4 firmware is still trying to merge the vision data with its own information. Is there a way to tell the PX4 to only use the data I am sending it? When I add fuse_brometer to the lpe_fuse paramter, then [lpe] vision position ok message comes out, but I want to fly UAV in indoor and precise data is required that I dont want to use barometer data. 

It looks like INAV mode is not support recently.. if I want to use INAV mode with Pixhawk 2.1, is thier recommend firmware version?

Thank you

TobiMiller, 2017-11-13:
for me everything works fine with EKF2 instead of LPE.

mandu46, 2017-11-13:
@TobiMiller Thank you for reply. Since I saw that this sentence "Indoor scenarios and general R&D applications continue to benefit greatly from the LPE estimator.", and my testbed is indoor environent, I wanna try with LPE. Or is accuracy of EKF2 enough for indoor enviroment at 1.6.3 firmware?

TobiMiller, 2017-11-13:
didnt saw that much of difference ;)

mandu46, 2017-11-13:
@TobiMiller I really appreciate to your answer. I will try with EKF2

michael1874888, 2018-01-22:
Hello!
Did you solve your problems?We face the same problem recently, and the problems were still there even we tried EKF2 mode. Do you have any idea?

mandu46, 2018-01-22:
Hi @michael1874888 . Unfortunatly, I don`t solve the problem yet... I just using INAV mode with Pixhawk 1 from now on.. I hope EKF2 solve the problem..but it looks not. If you solve the problem please let me know. Thank you

TSC21, 2018-01-23:
> When I add fuse_brometer to the lpe_fuse paramter, then [lpe] vision position ok message comes out, but I want to fly UAV in indoor and precise data is required that I dont want to use barometer data.

Just increase the std deviation for baro and the problem is solved. This is actually something already registered before and to be fixed.

closed 2018-01-23 by TSC21