PX4/PX4-Autopilot #8260: eph, epv, and evh thresholds should be larger for fixed wing

Antiheavy, 2017-11-09:
**eph_threshold**, **epv_threshold**, and **evh_threshold** should be increased for fixed wing to avoid unnecessary in-flight loss of global position due to commander rejecting it.  The hard coded values are more appropriate for multi-copter hovering, but are overly restrictive for fixed wing.

@priseborough @dagar , I'd appreciate your thoughts if this is appropriate.

https://github.com/PX4/Firmware/blob/9c8dc3941dbf56af20914b77a6ab585861c9afe4/src/modules/commander/commander.cpp#L199

static float eph_threshold = 5.0f;	// Horizontal position error threshold (m)
static float epv_threshold = 10.0f;	// Vertivcal position error threshold (m)
static float evh_threshold = 1.0f;	// Horizontal velocity error threshold (m)

I suggest an increase of 5X for fixed wing, this is just a guess really. 

Increasing these values is one approach that would have prevented the loss of global position failsafe in this log:  https://review.px4.io/plot_app?log=298fa694-516a-4d76-9d18-a5137adbc0d7

Other improvements in work related to the above log: https://github.com/PX4/ecl/issues/352

tdnet12434, 2017-11-13:
I also encounter the same issue when launching my fixed-wing with high power launcher. 
It seems ekf2 estimator back out and lost all control surface then crash. See my following log file 
https://review.px4.io/plot_app?log=ccc86ede-a1fa-4b92-afec-34987e5455e2

This procedure can be reproduced by using the catapult launch method to takeoff in gazebo plane simulation.
Then try to apply some force like 10,000 N (just impulse). It will give us no global position then the failsafe mode is suddenly activated.

dagar, 2017-11-16:
@tdnet12434 I believe the particular issue you hit (the EKF stops publishing) is resolved by this commit. https://github.com/PX4/ecl/commit/2e03084d349ce4ba404615f8c79b6d820d3335b6
Which version of PX4 was this?

dagar, 2018-01-28:
COM_HOME_H_T = eph threshold
    COM_HOME_V_T = epv threshold

evh is still only hardcoded

Sounds like we should add separate parameters for all 3 and maintain the separate dedicated HOME thresholds.

Antiheavy, 2018-04-24:
@dagar has this been address by: https://github.com/PX4/Firmware/pull/8771 ?

dagar, 2018-04-25:
Fixed in https://github.com/PX4/Firmware/pull/8771

closed 2018-04-25 by dagar