PX4/PX4-Autopilot #8269: Possible to Arm without GPS Lock

tops4u, 2017-11-10:
**Bug Report**

It is possible to Arm the Vehicle within about 20secs from Booting without having a valid GPS lock. This is most probably due to a check in the PrearmChecks that does not enforce GPS Check for EKF (when COM_ARM_WO_GPS is disabled) within 20secs after booting. 

It is possible to arm an takeoff during this timewindow without GPS Lock even when having Valid EKF Thresholds for GPS Signal and COM_AMR_WO_GPS set to false.

acfloria, 2017-11-13:
If GPS is required for arming is specified by the [COM_ARM_WO_GPS](https://docs.px4.io/en/advanced_config/parameter_reference.html) parameter.

tops4u, 2017-11-13:
Yes, I agree. But check the last segment of PreflightCheck.cpp

Line 778 defines that if there is no Lock, _time_last_no_gps_lock = time since boot when there is no lock yet. 

And 788 enforces GPS check within EKF Limits only when _time_last_no_gps_lock is > 20 secs. 

So in case there was no Lock and less than 20 secs after boot Arming is attempted there will be no Check if there is a GPS Lock. 

COM_ARM_WO_GPS did not change this behaviour in my test.

tops4u, 2017-11-17:
I did some tests with COM_ARM_WO_GPS set to FALSE

Here the Screenshot where I was able to ARM within the 20secs Window after Boot I was covering the GPS with a ESD Bag and my Hand (it lost the Lock), then rebooted the FMU and armed:
[image]

Messages in QGC:
[image]

After the 20sec Window is expired the PreflightCheck complains correctly, that there is no GPS available. BUT: It is not possible to recover from this, even when there is GPS available still there is the same Prearm Failure preventing to Arm. Only Reseting the FMU will allow arming. 
[image]

dagar, 2017-11-27:
Improvements here. https://github.com/PX4/Firmware/pull/8374

priseborough, 2017-11-28:
Having a GPS lock does not mean the vehicle will arm. The EKF has to have passed it's internal GPS quality checks and started using GPS. 

I will rework the 20 second startup check logic.

priseborough, 2017-11-28:
I've reworked the checks and pushed the fix to https://github.com/PX4/Firmware/pull/8374

tops4u, 2018-01-04:
Checked and looks ok for me. Thanks

closed 2018-01-04 by tops4u