PX4/PX4-Autopilot #8270: Global initialization of attitude Q-estimator heading

Seanmatthews, 2017-11-11:
This issue is not necessarily a bug, but elements of it, I have found, lead to decreased attitude estimation accuracy, as well as a pesky delay in the time it takes for the attitude estimator to "settle" upon supplying it with an external heading estimation (vision or mocap). 

The core of the problem stems from the forced initialization of the attitude estimator in the world frame, as opposed to the local body frame.  [Here](https://github.com/PX4/Firmware/blob/master/src/modules/attitude_estimator_q/attitude_estimator_q_main.cpp#L488-L490), the magnetometer is used to initialize heading relative to true north. [Above that](https://github.com/PX4/Firmware/blob/master/src/modules/attitude_estimator_q/attitude_estimator_q_main.cpp#L483-L486), a similar situation occurs with the accelerometer. 

In my case, I specify that 1) the magnetometer should not be used, and 2) an external heading from vision should be used. The init() method ignores both these requests by using the magnetometer, and by completing initialization before it's received any estimates from the external process. When an external pose in the body frame is finally supplied, the estimator jumps and takes, typically, 20-60s to zero the orientation coming back from the LPE, all while the vehicle is stationary. Furthermore, when the heading estimate does settle, it typically retains 1-2 degrees of starting error.

These problems led me to the idea that I could read the orientation values of attitude_estimator_q, before I supply my vision estimate, so that I may transform the estimate I supply to PX4 while still operating from a zeroed orientation on my end. However, I cannot access those values without first initializing the LPE with the vision estimates it expects from me (LPE_FUSION 12). 

The solution I finally went with was to alter attitude_estimator_q to initialize upon, and with, the first received vision pose, if the user has specified that an external vision-estimated heading should be used (ATT_EXT_HDG_M 1). This fix solves my problem, but I want to pose this problem to the community for recommendations on how to proceed. Should initialization in the body frame be an option? Or perhaps, should that extend to other custom frames?

dagar, 2017-11-12:
@jgoppert @TSC21 FYI

TSC21, 2017-11-17:
> Should initialization in the body frame be an option? Or perhaps, should that extend to other custom frames?

Yes to both. Do you have a proposal that can be extended to a code change that fits this purpose? I guess we should automatically adjust the source of data to the frame we are working with.

TSC21, 2017-11-17:
> Or should I go alter MAVROS to individually expose the values from attitude_estimator_q?

Not the ideal approach IMO.

closed 2018-02-21 by PX4BuildBot

Neoster, 2018-04-05:
@Seanmatthews has there been made any changes to the code regarding att_estimator_q initialisation when using mocap or vision? Did you solve your 'not necessarily' bug?