PX4/PX4-Autopilot #8271: A new mag calibration algorithm is need for reducing data storage heap

zarathustr, 2017-11-12:
The magnetometer calibration is based on a least square fitting of raw data. Currently, the algorithm requires a heap load of raw data storage arrays. This point significantly increases the online memory usage of the algorithm also increases instability of managing the RAM. A better issue is raised here to integrate my recent contributions into the PX4 that it no longer stores the raw data and uses a 10x10 matrix to compute the singular value decomposition results. Such method swaps the execution time with the space consumption, which is better for low-configuration platforms e.g. stm32.

dagar, 2017-11-12:
Let's talk about doing this after v1.7.0 goes out shortly. Calibration code will be moving out of commander (https://github.com/PX4/Firmware/pull/8200) and needs other improvements (https://github.com/PX4/Firmware/issues/8265) and fixes (https://github.com/PX4/Firmware/pull/8190).

zarathustr, 2017-11-23:
@dagar So when will we receive the 1.7.0 release? Then what are the main advances included in this release?

dagar, 2017-11-23:
We're in the process of testing and resolving a few remaining issues. https://github.com/PX4/Firmware/milestone/23

It's a huge list of things (everything that went into master over the last 4-5 months).

mhkabir, 2018-04-15:
@zarathustr Could you open a PR with your changes to the calibration routine?

closed 2018-08-06 by santiago3dr