PX4/PX4-Autopilot #8275: CI: broken Jenkins SITL

korigod, 2017-11-12:
Jenkins SITL is now broken. Based on console output I see two problems:

1. Github ids are now exceeding uint32 max value. It's known and fixed problem (https://github.com/kohsuke/github-api/issues/387), it may require Jenkins plugin upgrade.
2. Pandas tries to install new numpy version but still uses existing 1.8.2, which leads to import error and test failure. Sent PR #8274 to address it.

korigod, 2017-11-13:
Another problem is related to the SITL itself: I sometimes get "ACCELS INCONSISTENT". It happens in Jenkins too, e.g. [this Jenkins build](http://sitl01.dronetest.io/job/PX4%20Firmware%20POSIX%20SITL/2104). That commit was built twice as PR and it passed all tests.
Here is the log: https://logs.px4.io/plot_app?log=0a5956f8-a1e4-4a15-8908-13bf63586373

dagar, 2017-11-13:
I've been wanting to revamp this entire setup and get it moved to a new jenkins cluster. http://ci.px4.io:8080/blue/organizations/jenkins/

dagar, 2018-01-22:
Fixed.

closed 2018-01-22 by dagar