PX4/PX4-Autopilot #8277: Replace astyle with clang-format

hamishwillee, 2017-11-13:
Astyle version updates have historically caused problems with large diffs etc. Proposal from @bkueng was that this might best be replaced by clang-format.

Discussion here: https://github.com/PX4/Devguide/pull/331#issuecomment-343414776

bkueng, 2017-11-13:
In addition we could use [clang-tidy](https://sarcasm.github.io/notes/dev/clang-tidy.html) to check the coding style.
I ran some tests using an as similar style as we already use. This is the automatic transformation diff: https://github.com/bkueng/Firmware/commit/0efb7e0083e993b9bccaa1267798ac838326caf2. The result does not compile, but requires additional manual steps. There are 87 fatal build errors out of 560 steps, which gives a rough estimate how much manual intervention is needed. Also I used clang 4.0.0, maybe a newer version can do better.

korigod, 2017-11-13:
The question is do we want to switch to camel case? I believe underscores are totally acceptable in C++, e.g. they are used in STL.

bkueng, 2017-11-13:
> The question is do we want to switch to camel case? I believe underscores are totally acceptable in C++, e.g. they are used in STL.

I think so too, and the actual style is of course up to discussion. I tried to match it, but it's already inconsistent. Consistency is generally more important to me, as it makes the code easier to read and understand.

LorenzMeier, 2017-11-13:
No, we don’t want to switch to camel case. We are roughly following the Google style guide with a few deviations.

dagar, 2017-11-14:
There are two separate issues - 

 1. Astyle version problems. I would personally like switching to clang-format for other reasons, but I don't think it's going to remove all the versioning pain. This might be obnoxious, but one idea is to have the CI system push formatting fixes to your PR instead of just complaining.

2. Using clang-tidy for consistent naming. I'm all for this if we can do it incrementally.

dagar, 2017-11-19:
Issue to discuss coding style in general - https://github.com/PX4/Firmware/issues/8319

closed 2018-02-21 by PX4BuildBot