PX4/PX4-Autopilot #8278: Offboard revision

Stifael, 2017-11-13:
PX4 offboard is part of PX4 that is used by many people, but when looking through the code, it becomes evident that it grew over time quite a lot without having a very clear software architectural plan. Here are some points that I would like to discuss:

- Triplets
https://github.com/blog/1375-task-lists-in-gfm-issues-pulls-comments
At least we should use position setpoint msg [here](https://github.com/PX4/Firmware/blob/master/msg/position_setpoint.msg) instead of the triplet msg. But even the single position setpoint msg is rather large, with stuff in there that is not and never should be part of offboard.

- Takeoff, land, loiter...
Offboard supports besides of position, velocity and acceleration setpoints also auto related features such as takeoff, land, loiter and so on ([here](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L922-L935). In addition, ppl also expect to have offboard waypoint maneuvers  (for instance [here](https://github.com/PX4/Firmware/pull/8267#issuecomment-343873609)). 
To me, offboard should be simple. It should provide an interface where users can program their own takeoff, land and loiter within an offboard module exploiting the postion, velocity and acceleration setpoint (loiter is anyway just sending the same position setpoint continuously). Auto related features can be offered, but should not have priority. In addition, the distinction between offboard initiated auto features to raw position setpoints should be clear. For instance, if the decision is to support waypoint maneuvers, then there should be a clear distinction to the standard position setpoints used for instance to follow a path. The reason for doing that is make it clear to the offboard developer what he can expect. For instance, if the offboard developer implements a L1 controller, he expects that the position setpoints correspond exactly to what was sent over offboard with the tightest velocity controlller possible. Any farther higher level setpoint adjustment such as slewrate would be confusing. On the other hand, if the user demands a waypoint maneuver, which is the same as auto, then the already implemented auto logic can be used instead of writing another offboard logic for it. 

- current offboard slewrate problem
the overshoot offboard problem [here](https://github.com/PX4/Firmware/pull/8240) is the same issue as we had in auto and manual (https://github.com/PX4/Firmware/issues/6681), where direction change can cause huge overshoots. This means that if we want to support waypoint manuevers, I suggest to use the current implemented auto logic. An other possibility is to reset the slewrate similar to what is done in manual, but I don't recommend that.

Stifael, 2017-11-13:
@korigod your opinion is very much appreciated.

MaEtUgR, 2017-11-14:
> At least we should use position setpoint msg [here](https://github.com/PX4/Firmware/blob/master/msg/position_setpoint.msg) instead of the triplet msg.

What about the clean [local position setpoint](https://github.com/PX4/Firmware/blob/master/msg/vehicle_local_position_setpoint.msg) and also offer a global one for the offboard user (but convert it for the controller on reception)?

> On the other hand, if the user demands a waypoint maneuver, which is the same as auto, then the already implemented auto logic can be used instead of writing another offboard logic for it.

I think this is an important point you made. If you want to to exactly have what auto does, just give commands to auto features over mavlink. No need to support the same automatic takeoff or waypoint line over the offboard inteface again.

> current offboard slewrate problem
Worst heavy overshoot most likely quick fixed by #8282 (please confirm) and after that I would either disable slewrate for offboard because it anyways degrades performance or if it's desired revise it to not have these problems when the setpoint can not be tracked (will probably take time).

AlexisTM, 2017-11-15:
You should join the channel Architecture on the PX4 Slack 
@jones

closed 2018-02-21 by PX4BuildBot

MaEtUgR, 2018-02-21:
@Stifael I guess this could be realized in an offboard flight task ... #8874