PX4/PX4-Autopilot #8280: krait update param failed

ChristophTobler, 2017-11-13:
I tried running PX4 on a Snapdragon Flight (`make eagle_legacy_default`). It builds using https://github.com/PX4/Firmware/pull/8279.

When I run `./px4 mainapp.config`, I get
http://pasted.co/4034af78
mini-dm
http://pasted.co/05edbbf5

ChristophTobler, 2017-11-13:
similar to https://github.com/PX4/Firmware/issues/7364. There the aDSP crashed, right?

dagar, 2017-11-13:
@darioxz I think we need to get one of these added to the CI system. It's way too easy to silently break.

ChristophTobler, 2017-11-14:
There is no error when using `eagle_default`.

FYI @mcharleb

ChristophTobler, 2017-11-15:
v1.6.5 is still working

LorenzMeier, 2017-11-15:
@ChristophTobler This is likely the result of parameters being present on the Linux side that are not present on DSP or vice-versa. @dagar this is one of the reasons why I'm not a big fan of "cleanups" as long as we've not solidified the underlying architecture to take care of "enabled parameters" in a fundamental fashion.

LorenzMeier, 2017-11-15:
@ChristophTobler Please try a commit from 3 weeks ago and work your way in.

dagar, 2017-11-17:
> @dagar this is one of the reasons why I'm not a big fan of "cleanups" as long as we've not solidified the underlying architecture to take care of "enabled parameters" in a fundamental fashion.

I don't think this is related to enabled parameters (and that big PR was never merged), but I see your point.

I'm wondering if it's something with the parameter culling combined with out of source modules (FC_ADDON)? @ChristophTobler can you diff the parameters.xml in each build (posix + qurt)?

ChristophTobler, 2017-11-22:
fixed in https://github.com/PX4/Firmware/pull/8325

closed 2017-11-22 by ChristophTobler