PX4/PX4-Autopilot #8295: LPE with External Vision Position crash into ceiling while in Altitude Control

Ekci, 2017-11-15:
Logs here: [https://logs.px4.io/plot_app?log=d0e992b7-5855-45eb-87a4-2bd08147d201](link)

I am currently working on indoor navigating using External Vision Position (mavros). Normal flight looks like this: I arm, take off, hover at around 1 meter at which points my camera starts to report position, change to alt hold and then to offboard which engages navigation to point (0,0) at 1.8 meters.

We had few flights without any problem (we are still tuning) but last flight ended with crashing into ceiling after switching to alt hold. Question is: what LPE altitude control is using as alt reading? Why we were able to fly in alt hold without any problem (even when we were loosing external position)?

I realized that my local position estimator was not "following" external one but why it was ignoring my baro and range sensor?

TobiMiller, 2017-11-16:
Use EKF2. i know in docs it is recommend to use LPE but i think it is outdated. change the parameter for altitude fusion from baro to vision. if it dosnt help. try to send a setpoint and set the flightmode to offboard. i experienced a similar problem with pos. hold and thought that this flightmodes only works with gps

TSC21, 2017-11-16:
@TobiMiller why do you say it is outdated? What it really needs is proper tuning

TobiMiller, 2017-11-16:
@TSC21 it is recommend to use LPE. out of the box EKF2 worked better for me. thats why i thought this recommendation is maybe from times before ekf2 ;)

TSC21, 2017-11-16:
Befor ekf2 had vision support

Ekci, 2017-11-16:
I'm confused, what's better to use now? LPE or EKF2?
Anyway, I was able to stabilize indoors for around 30 minuts using LPE. My problem is last crash when changing to alt hold...

TSC21, 2017-11-16:
Set `LPE_FUSION` bitmask so it can also fuse baro. Then set `LPE_BAR_Z` to an high value.

TSC21, 2017-11-16:
The problem was the Z estimation.

Ekci, 2017-11-16:
Yeah, that's wat I was thinking. What about Lidar? Is there a way to fuse it?

TSC21, 2017-11-16:
It should automatically fuse. You don't need to set a parameter.

Ekci, 2017-11-16:
Ok, so why the Z estimate was that off if Lidar was on and even External Vision Position turned on before switching to alt hold? You can clearly see that Vision Position Z was around 1.5 meters when we change modes. I agree that we should have given it more time to fuse but I was sure that Z will be correct because of the Lidar.

TSC21, 2017-11-16:
Try setting `pub agl as lpos down` on `LPE_FUSION`.

TSC21, 2017-11-16:
Also try to decrease `LPE_VIS_Z` (Vision z standard deviation)

Ekci, 2017-11-16:
Ok, I will try it next time. Thank you for helping here, I will close this issue for now.

closed 2017-11-16 by Ekci