PX4/PX4-Autopilot #8296: Sporadic state switch to AUTO.LAND

Seanmatthews, 2017-11-15:
Snapdragon Flight standard setup
PX4 v1.6.5
MAVROS 0.19.0

I'm running px4 with local_position_estimator, attitude_estimator_q, and land_detector (multicopter). An RC receiver is not connected, nor do I run the rc_receiver module. PX4 starts in MANUAL mode, and I do not try to change that. The piece of code I'm running simply arms and disarms the vehicle, at intervals, from a landing state. Somewhat randomly, the FC state changes to AUTO.LAND, causing the thrusters to spin up briefly but violently. I'm sending no other inputs to either PX4 or MAVROS. I had been playing around with the COM_DISARM_LAND parameter, but I see the same behavior whether it's 0 or greater than.

Is this normal? Is there a state I can command that instructs PX4 to remain in an "idle" mode, where it is incapable of switching to AUTO.LAND? Essentially, I want to be able to send [dis]arming commands to PX4 without fear that the mode will automatically switch on me, even with estimator and land_detector inputs.

dagar, 2017-11-16:
MANUAL mode with no RC + armed + flying = switch to AUTO LAND as a failsafe.

I'd need to see the log to say more, but it's also likely due to the land detector tuning.

Seanmatthews, 2017-11-16:
@dagar Thanks. I'll work on getting a log. To make it clear, I am not flying. What typically triggers the failsafe? Is it a deduced sensory input such as a takeoff or land detection? I looked into the state machine code briefly, but it's not immediately clear if that's the case.

dagar, 2017-11-17:
By flying I just mean not landed (as reported by the land detector). The log will tell us for sure.

closed 2018-02-21 by PX4BuildBot