PX4/PX4-Autopilot #8297: Error compiling v1.7.0-rc1 for eagle_default target

r0gelion, 2017-11-15:
I'm using Ubuntu 14.04LTS

1. I tried to compile v1.7.0-rc1:
`make eagle_default`
2. Got the error: http://pasted.co/0a7d6f31
2.1. This error recommends running the next command:
`sudo apt-get install python-toml python-numpy`
2.2. I ran the previous command but I received the error that python-toml package wasn't located.

I follow the next steps to to solve the issue:
1. Downloaded the python-toml deb file from [here](https://packages.ubuntu.com/xenial/all/python-toml/download)
2. Installed the deb package using Ubuntu's package installer.
3. Ran the next command again:
`sudo apt-get install python-toml python-numpy`
After the above step I was able to compile

@jlecoeur @dagar

sparky002, 2017-11-16:
I ran into the same problem compiling v1.7.0-rc1 for px4fmu-v2_default and px4fmu-v3_default on ubuntu 14.04lts. Same solution worked on my end.

huangwen0907, 2017-11-21:
@r0gelion 
>   sudo -H pip install pyserial empy toml pandas jinja2
Maybe can fix you problem.

r0gelion, 2017-11-22:
@huangwen0907 Thank you, the problem was solved with the process shown at the issue description.
 
@jlecoeur should we keep this issue open?

closed 2017-11-26 by dagar

anandbaskaran, 2018-04-20:
I am still getting this same error. I tried installing with both pip and with apt-get and it installed properly. But, whenever I run the make of PX4 I get the same error.

hamishwillee, 2018-04-23:
@anandbaskaran Best not to hijack issues that have been closed some time ago. Create a new one describing the environment you used and the precise problem.

PR #8338 (2017-11-26): Python import error: Recommend pip over apt-get
closes #8297

The package python-toml does not exist in older (<= 14.04) versions of ubuntu.
commit 9439eaa5 (2017-11-26): Python import error: Recommend pip over apt-get

closes #8297

The package python-toml does not exist in older (<= 14.04) versions of ubuntu.