PX4/PX4-Autopilot #8299: F450 frame with Pixhawk mini crash - v1.7.0-rc1

r0gelion, 2017-11-16:
Using F450 frame with Pixhawk mini autopilot, loaded v1.7.0-rc1

The vehicle was flying in mission mode and when going to the second waypoint it stopped to follow the mission, it didn't respond to RTL command (by RC) and it crashed.

https://logs.px4.io/plot_app?log=57cd3193-b22d-478a-8c10-c0ff0f9a5178

@dagar @LorenzMeier

dagar, 2017-11-16:
What was the crash like? Did the battery get so low the ESCs finally shut off? Do these numbers look believable?

[image]

r0gelion, 2017-11-16:
The ESC's didn't cut, the vehicle flew to a side and crashed. The battery got hit and a cell is dead but cannot assure if it died before or after the crash.

dagar, 2017-11-16:
Oh, I see the flight was upfront.

[image]

The red line is the mode. It starts in position control, the mission, and is already in RTL before you lost RC.

dagar, 2017-11-16:
I still think it's a battery issue where it was so low it couldn't maintain altitude during the RTL and crashed. Until it's unable to maintain it was on a straight path back to home.

The RTL was triggered by low battery.
[image]

[image]

Shortly after 2:40 you start losing altitude during the RTL and the impact happens at about 2:50. The battery flat lines, even though the commanded thrust goes to maximum.

[image]

dagar, 2017-11-16:
You should change the low battery action (COM_LOW_BAT_ACT) to 3 "Return to land at critically low level, land at current position if reaching dangerously low levels". 

At the critical threshold (default 7%) the vehicle will do an emergency land instead of the RTL.

r0gelion, 2017-11-17:
We will follow your recommendation.

I'm closing this issue.

Thanks!

closed 2017-11-17 by r0gelion