PX4/PX4-Autopilot #8305: Solo PREFLIGHT FAIL: MAG SENSORS INCONSISTENT

kandai-wata, 2017-11-17:
Hello,
I'm not sure if this could be an issue report, but I will post the problem here as I could not find any info on PX4Disscuss, or on other github repositories.

I am now working on flying 3DR Solo indoor with VICON for research.
As I googled and came to a conclusion that PX4 is the solution to incorporate VICON data using MavRos, and not Ardupilot (I hope this is true).  
So I flashed PX4 with `make px4fmu-v3_default`, as lpe version could not be found for v3 (Pixhawk2). 

The problem I got was not with mocap, but with magnetometer. 
Even If I calibrate, I always get `PREFLIGHT FAIL: MAG SENSORS INCONSISTENT` on qgroundcontrol.
According to [PX4 Tutorial](https://dev.px4.io/en/tutorials/pre_flight_checks.html), sensors inconsistent means it needs to calibrate, but I tried few times.
I am suspecting the orientation setting. As I searched on the web, Solo has HMC 5983, LSM303D onboard, and HMC 5983 for external magnetometer, and the orientation of onboard sensors is `pitch 180` and that of external is `Yaw293Pitch68Roll90`, however `Yaw293Pitch68Roll90` is not on the list in qgroundcontrol (please see an image below). 
As `Yaw293Pitch68Roll90` was prepared in `src/lib/conversion/rotation.cpp`, I thought by setting `CAL_MAG0_ROT` to 30 would work, but not successful. 
I am also wondering why qgc is telling me that there are 4 magnetomers, external: 0 & 1, internal 2, 3 (please see an image below) 
I am assuming that the orientation setting of external mag is wrong, so that is why qgc is giving me that error. 
Another reason might be the calibrated radio. I am using 3DR Solo's controller but calibration cannot be completed with this controller.

I also tried to flash the firmware via USB cable, but I got same result.
Is there any ways to fix this error or disabling external mag? I can use the orientation data from VICON so  magnetometer is not really needed in my opinion. 

The only person who successfully flashing px4 on Solo that I could find was @lovettchris . 
I hope I'm not ruining this repository.... Please tell me if I am. 
I am very new to pixhawk, px4, mavros and etc., but I am doing my best to fly my drone! 
It would be a great help If you could help me with this.

[image]
[image]

In addition, this may be off topic, but could someone check if my procedure for flying px4 indoor is right? 
On PX4:
 - use_tf =true (in the px4_config.yaml file)
 - SYSTEM_COMPANIN to 921600 
 - ATT_EXT_HDG_M = 2 (motion capture)
 - LPE_FUSION = 12

On Offboard PC (Ubutu 14.04 LTS):
1. Run `roslaunch mavros px4.launch fcu_url:=udp://0.0.0.0:14550@10.1.1.10:14555`
2. Run the similar [example](https://dev.px4.io/en/ros/mavros_offboard.html) written in the tutorial,
but also including
- publisher that publishes VICON data to the topic /mavros/mocap/pose or to /mavros/vision_pose/pose
2.2 
 - sending `setpoint_position` to activate OFFBOARD

dagar, 2017-11-17:
The mag inconsistency check was only recently added, and from the sounds of it in this case it's catching that one of the mags has the wrong rotation.

https://github.com/PX4/Firmware/issues/7696

kandai-wata, 2017-11-17:
Thank you for your reply. 
Does that mean I should use the master older than #7696 ?
The magnetometer is fully equipped with Solo, so this is not a hardware problem, but a software problem catching wrong rotation??

dagar, 2017-11-17:
It's a software bug on this particular hardware. You can increase COM_ARM_MAG to get around it, but know that one of the backup mags is actually incorrect. If the primary failed, the backup is worse than nothing. It's still present in older firmware, but slips through unnoticed without the check.

To verify this is definitely the issue you could create a flight log pointing the vehicle north and rotation 360 degrees. Set SDLOG_MODE 2 to log immediately and SDLOG_PROFILE enable "sensor comparison". Then we can look at all the mag data side-by-side and easily see the problem.

https://dev.px4.io/en/advanced/parameter_reference.html#sd-logging

dagar, 2017-11-17:
This pull request should have the proper fix. https://github.com/PX4/Firmware/pull/8300

LorenzMeier, 2017-11-18:
I'm closing this - please comment in the fix PR.

closed 2017-11-18 by LorenzMeier

ofreedream, 2017-11-19:
nice