PX4/PX4-Autopilot #8307: Feature Request

PujieXin, 2017-11-17:
*Feature Request*
is it possible to simulate multiple drones and vehicles(ArduRovers) in Gazebo Simulation?Which part should I modify? The multi vehicle launch file?https://github.com/PX4/Firmware/blob/master/launch/multi_uav_mavros_sitl.launch
Thank you for help in advance.

ndepal, 2017-11-17:
Have you seen this?
https://dev.px4.io/en/simulation/multi-vehicle-simulation.html

PujieXin, 2017-11-17:
Yes， I check the multi_uav_mavros_sitl.launch file, I found there are 2 uav models loaded in this file. I mean to change one of the uav into ArduRover. Do you think this will be possible?  @ndepal

TSC21, 2017-11-17:
> I mean to change one of the uav into ArduRover.

Unless you add the ArduRover model into `sitl_gazebo` and use the PX4 rover capabilities, that won't be possible. You may want to look for this on the SITL capabilities of Ardupilot instead.

TSC21, 2018-01-05:
@PujieXin what's the status?

closed 2018-08-06 by santiago3dr