PX4/PX4-Autopilot #8312: 3 mixer sources ch 5,6 completely stop other pwm outs ch 7 and 8

taileron, 2017-11-18:
I tried the following mixer (Convergence Vtol) for adding a tilt factor controlled by aux 3 on a PixFalcon with 1.7 rc 0 and 1 and as soon as there are 3 instead of 2 mixer sources (M: 3) the elevon outputs stop to work.
The tilt mechanism shows the desired behaviour in all cases and mixes all 3 souces in the right way. (it worked with the master 1.6 from august 17)

PWM info displays 0 us and no pwm signal on output 7 an 8 (elevons)

# Motors
R: 3y 10000 10000 10000 0

Z: 

Tilt mechanism servo mixer
---------------------------
#RIGHT up:2000 down:1000
M: 3
O:      10000  10000      0 -10000  10000
S: 1 4   4000 -13200   7750 -10000  10000
S: 0 2   8000   8000      0 -10000  10000
S: 3 7  -9000  -9000      0 -10000  10000

#LEFT up:1000 down:2000
M: 3
O:      10000  10000      0 -10000  10000
S: 1 4   4000  13600  -5450 -10000  10000
S: 0 2   8000   8000      0 -10000  10000
S: 3 7   9000   9000      0 -10000  10000

Elevon mixers
-------------
M: 2
O:      10000  10000      0 -10000  10000
S: 1 0  -7500  -7500   1215 -10000  10000
S: 1 1   8000   8000      0 -10000  10000

M: 2
O:      10000  10000      0 -10000  10000
S: 1 0  -7500  -7500   1215 -10000  10000
S: 1 1  -8000  -8000      0 -10000  10000

tubeme, 2017-11-20:
I had similar experience, but with another aspect of the mixer. If I use the AETRFG mixer and I want to use the rudder source in more than one channel then the rudder stop responding. The problem was that we had a wheel steering  servo and the and the control group 0 7 did not move the servo at all. So we tried control group 0 2 for the purpose but it did not work as well.

bresch, 2017-11-23:
@taileron Do you have everything on the MAIN outputs? Could you try to use AUX for the servos?

taileron, 2017-11-23:
@bresch the strange thing is that all others work properly and as soon as the third mixer source of ch 5,6 is removed ch 7,8 start to work again

taileron, 2017-12-09:
found a very similar issues that had been finally solved a year ago #5924

taileron, 2017-12-12:
with the current master mixer of ch 7 sometimes starts:
HW arch: PX4FMU_V2
HW type: V2
HW version: 0x00090008
HW revision: 0x00000000
FW git-hash: 3c667bb4f5e48da54408fa4890a2872b08fa63af
FW version: 1.7.0 0 (17235968)
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: b18053574bf41712cef93e31bf178518f451a350
Build datetime: Dec 12 2017 17:13:56
Build uri: localhost
Toolchain: GNU GCC, 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
MFGUID: 31323634343551160049002d
MCU: STM32F42x, rev. 3
UID: 49002D:34355116:31323634 
device: /dev/pwm_output0
channel 1: 900 us (alternative rate: 400 Hz failsafe: 900, disarmed: 900 us, min: 1080 us, max: 1943 us, trim:  0.00)
channel 2: 900 us (alternative rate: 400 Hz failsafe: 900, disarmed: 900 us, min: 1080 us, max: 1943 us, trim:  0.00)
channel 3: 900 us (alternative rate: 400 Hz failsafe: 900, disarmed: 900 us, min: 1080 us, max: 1943 us, trim:  0.00)
channel 4: 1493 us (alternative rate: 400 Hz failsafe: 1925, disarmed: 900 us, min: 976 us, max: 2011 us, trim:  0.00)
channel 5: 1985 us (default rate: 50 Hz failsafe: 1985, disarmed: 0 us, min: 650 us, max: 2450 us, trim:  0.00)
channel 6: 1030 us (default rate: 50 Hz failsafe: 1030, disarmed: 0 us, min: 650 us, max: 2450 us, trim:  0.00)
channel 7: 1519 us (default rate: 50 Hz failsafe: 1520, disarmed: 0 us, min: 820 us, max: 2220 us, trim:  0.00)
channel 8: 0 us (default rate: 50 Hz failsafe: 1520, disarmed: 0 us, min: 820 us, max: 2220 us, trim:  0.00)
channel group 0: channels 1 2
channel group 1: channels 5 6 7 8
channel group 2: channels 3 4
nsh> px4io status
WARN  [px4io] loaded
protocol 4 hardware 2 bootloader 3 buffer 64B crc 0x66a7822b
8 controls 8 actuators 18 R/C inputs 2 analog inputs 0 relays
1064 bytes free
status 0x3741 OUTPUTS_ARMED SAFETY_OFF RC_FAIL FMU_OK MIXER_OK ARM_SYNC INIT_OK
alarms 0x0030 FMU_LOST RC_LOST
vservo 0 mV vservo scale 10000
vrssi 1367
actuators 0 0 0 0 4840 -5770 -1 0
servos 900 900 900 1493 1985 1030 1519 0
reversed outputs: [________] trims: r:   0.0000 p:   0.0000 y:   0.0000
0 raw R/C inputs
R/C flags: 0x0000
mapped R/C inputs 0x0000
ADC inputs 0 1591
features 0x0008 RSSI_ADC
arming 0x0439 FMU_DISARMED IO_ARM_OK FAILSAFE_CUSTOM INAIR_RESTART_OK ALWAYS_PWM_ENABLE OVERRIDE_IMMEDIATE
rates 0x000f default 50 alt 400 sbus 72
debuglevel 0
controls 0: -4 -1 -2 0 0 0 0 0
controls 1: 0 0 0 0 0 0 0 0
controls 2: 0 0 0 0 0 0 0 0
controls 3: 0 0 0 0 0 0 0 0
input 0 min 915 center 1500 max 2082 deadzone 10 assigned 0 options 0x0003 ENABLED REVERSED
input 1 min 900 center 1500 max 2096 deadzone 10 assigned 1 options 0x0001 ENABLED
input 2 min 1031 center 1031 max 2101 deadzone 10 assigned 3 options 0x0001 ENABLED
input 3 min 908 center 1500 max 2094 deadzone 10 assigned 2 options 0x0003 ENABLED REVERSED
input 4 min 1000 center 1525 max 2000 deadzone 10 assigned 255 options 0x0000
input 5 min 1039 center 1500 max 1961 deadzone 10 assigned 255 options 0x0000
input 6 min 899 center 1410 max 1921 deadzone 10 assigned 5 options 0x0001 ENABLED
input 7 min 1000 center 1500 max 2000 deadzone 10 assigned 255 options 0x0000
input 8 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 9 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0002 REVERSED
input 10 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0002 REVERSED
input 11 min 1099 center 1500 max 1901 deadzone 0 assigned 7 options 0x0001 ENABLED
input 12 min 1099 center 1500 max 1901 deadzone 0 assigned 255 options 0x0000
input 13 min 1099 center 1500 max 1901 deadzone 0 assigned 255 options 0x0000
input 14 min 1099 center 1500 max 1901 deadzone 0 assigned 255 options 0x0000
input 15 min 1099 center 1500 max 1901 deadzone 0 assigned 255 options 0x0000
input 16 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 17 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
failsafe 900 900 900 1925 1985 1030 1520 1520
disarmed values 900 900 900 900 0 0 0 0

taileron, 2018-01-05:
How can I do a git clone https://github.com/PX4/Firmware.git to bring an 1.6.5 version to build an old code for further flight improvement of my tilt mixer ?

bresch, 2018-01-05:
Did you try the latest v1.7 release? Is it still broken?
If you want to checkout an older release you can `git tag` to list the tags and then `git checkout tags/<tag_name>` or even better, `git checkout tags/<tag_name> -b <branch_name>` to create a new branch starting at this tag.
For example: `git checkout tags/v1.5.0 -b dev-tilt-mixer`

taileron, 2018-01-05:
current 1.7.3 master is still broken, but build from old tag 1.6.5 branch works perfect. Is there an easy way to include commits in between tag 1.6.5 and 1.7.0 to determine a later status at which the mixer still worked?

taileron, 2018-06-18:
... changing source file: modules/pxiofirmware/mixer.h
line 42 from 230 to 330 will load mixers with more than 2 sources
On PixFalcon for me that worked perfect.
230 was a good value in V 1.6.5 but in versions above 1.7.x it is necessary to increase it
same here #8975

taileron, 2018-06-19:
... Release 1.8.0 broke it again.
with master 180dev from 10´th of june it worked perfect all mixer always loaded even with 5 sources (for vectored thrust)

taileron, 2018-06-19:
1.8.0 10´th june master supports even more sources than 1.6.5 now this mixer always starts:

# Motors
R: 3y 10000 10000 10000 0
# XFade Source (AUX1/RC7) MC=2000 FW=1000
M: 1 
O:      10000  10000      0 -10000  10000
S: 3 5      0 -19888   9808  -10000  10000
#S: 3 5      0  19308  -9508 -10000  10000

Tilt mechanism servo mixer + (Tilt AUX3/RC12) + (FW Pi/Roll)
---------------------------
#RIGHT up:1986 down:1030 Limit:650-2450
M: 5
O:      10000  10000      0 -10000  10000
S: 1 4   1000 -10610   4840 -10000  10000
S: 0 2   4444   4444      0 -10000  10000
S: 3 7  -5850  -5850      0 -10000  10000
S: 1 0   3000   3000      0 -10000  10000
S: 1 1  -3000  -3000      0 -10000  10000

#LEFT up:1030 down:2015 Limit:650-2450 Center=1550
M: 5
O:      10000  10000      0 -10000  10000
S: 1 4   1000  10940  -5770 -10000  10000
S: 0 2   4444   4444      0 -10000  10000
S: 3 7   5850   5850      0 -10000  10000
S: 1 0   3000   3000      0 -10000  10000
S: 1 1   3000   3000      0 -10000  10000

Elevon mixers Limit:820-2220 Center=1520
-------------
M: 2
O:      10000  10000      0 -10000  10000
S: 1 0  -7500  -7500      0 -10000  10000
S: 1 1   8000   8000      0 -10000  10000

M: 2
O:      10000  10000      0 -10000  10000
S: 1 0  -7500  -7500      0 -10000  10000
S: 1 1  -8000  -8000      0 -10000  10000

taileron, 2018-06-19:
.... sorry, 1.8.0 Final release works with line 42 changed to 330 (GCC 7.2.1)

closed 2018-08-06 by santiago3dr