PX4/PX4-Autopilot #8318: PX4 SD card corrupted after a crash

myeabhishek, 2017-11-19:
**Bug Report** or **Feature Request**

For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](slack.px4.io). Issue reports need to contain the items below:

 - Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/flight-reporting.html))
 - Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/)
 - Expected behavior and actual behavior.
 - Steps to reproduce the problem.
 - Specifications like the version of the project, operating system, or hardware.

Hi Team,

I have a multicopter with v1.6.5 stable release firmware. My drone flipped and crashed while landing in manual mode. On early analysis it was found that the SD card got corrupt and we were not able to retrieve any logs from it. 

Could it be because of issues like #8105 #7801 #7805 #7766 #7844 ?
What could be the possible reasons for SD card getting corrupt?

I wasn't using a gimbal but vmount application was running in AUTO in and AUX out configuration.

dagar, 2017-11-19:
It's hard to say for sure without having data to review. How was the flight prior to landing? Can you elaborate on the crash? Did it flip over at touch down or while still in the air? Do you have other successful flights with this configuration (specific vehicle, autopilot, firmware revision)?

myeabhishek, 2017-11-19:
Manual flight before landing was stable and while landing the drone suddenly flipped and fell on the ground. It flipped in the air itself about 2m above the ground. 

Yes I had successful manual flights before. Also we had successfully tested auto takeoff and hold as well. 

Following are the details regarding the vehicle:

Firmware version: 1.6.5 stable release
Vehicle: Quad X

dagar, 2017-11-19:
Did you look at the sdcard in a computer or through QGC? Is there anything at all on the card?

QGC should be saving telemetry logs to disk. If you can find the corresponding log it might have a clue.

tdnet12434, 2017-11-20:
I always suggest everyone using px4 on pixhawk to use at least class 10 sd card.
Because of cheap class 4 china sd card that already placed inside pixhawk, sometimes easy to corrupt log file and prone make serious problem like this.

Which sd card are you using? I doubt, maybe writing rate is issue?

myeabhishek, 2017-11-20:
@dagar The card is not mounting on windows, linux or mac. It needs to be formatted. I checked the microsd folder via ssh, there is a blank logs folder. The logs are not getting generated on this card now by even "Logger on" command. 

@tdnet12434 Yes I am using the sd card that came with my pixhawk.

arun2204, 2017-11-20:
I am attaching the tlog file from Qgroundcontrol. I replayed this file and towards the end when my it crashed there was no data being received from the drone and then after the crash it tried to takeoff again.

Could it be that my pixhawk hanged as reported by people in the mentioned issues like #8105 and #7801 because of vmount?

[crash_log.tlog.zip](https://github.com/PX4/Firmware/files/1487205/crash_log.tlog.zip)

myeabhishek, 2017-11-20:
I am attaching the tlog file from Qgroundcontrol. I replayed this file and towards the end when my it crashed there was no data being received from the drone and then after the crash it tried to takeoff again.

Could it be that my pixhawk hanged as reported by people in the mentioned issues like #8105 and #7801 because of vmount?

[crash_log.tlog.zip](https://github.com/PX4/Firmware/files/1487217/crash_log.tlog.zip)

tubeme, 2017-11-20:
@myeabhishek You could try to recover the card with HDD recovery tools if you cannot see any files on the card. If it not physically broken most likely you will succeed the recover the logs. The card is FAT32 or ExFAT so there is nothing special about the file system. It might be due to the broken MBR or something consequence from the sudden stop of power in the crash. Once you recover the files though the last LOG, the one from the crash might be corrupt as well and you could not be able to open it with any of the tools available. The reason for this is that the log file is not closed properly and the end of the log gets messy and corrupt. So you have to open the file with TXT editor and delete some last bits from it. I recovered a lot of log files this way after crashes.

Crashes are putting a lot of Gs on the card so we've had a couple of cases with physically broken cards as well. It depends on your luck.

closed 2018-02-21 by PX4BuildBot