PX4/PX4-Autopilot #8319: PX4 coding style guidelines + tooling

dagar, 2017-11-19:
Consider establishing PX4 coding style guidelines across all projects.

 - C++ (and python?) coding standards and style guidelines consistent across all PX4 projects
 - use clang-tidy to fix inconsistencies and gradually enforce
 - flight code should consider portions of HIC++ (http://www.codingstandard.com/section/index/) and MISRA C++

We need to be careful to not make it too difficult for new contributors, but establishing a level of consistency will make it easier to transition to working on different areas.

https://github.com/PX4/Firmware/issues/8277

dagar, 2017-11-19:
@bkueng @LorenzMeier @hamishwillee

hamishwillee, 2017-11-19:
@julianoes @mrpollo I'd say this is a Dronecode level thingy, not just PX4.

julianoes, 2017-11-20:
So for Python I would vote to use [PEP8](https://www.python.org/dev/peps/pep-0008/).

While I agree that it would be nice to have all C++ code under the DroneCode umbrella consistent, I have to say that the current Firmware style is quite different from most projects around it (or included in it). This refers mostly to the style and white space, and less to standards and practices which are more aligned in my opinion.

mrpollo, 2017-11-20:
Agree with @julianoes, on python we need enforce [PEP8](https://www.python.org/dev/peps/pep-0008/)

This is a tricky one to enforce, are we going to clean up the codebase on a single pass? whats the plan?

julianoes, 2017-11-20:
For PEP8? I'd say file by file. I've done some PEP8ifying in the past every now and then when I touched a file.

mrpollo, 2017-11-20:
Not just for PEP8 for the C++ code as well